cra-ros-pkg/robot_localization #153: Option to future date map->odom transformation

IvanVN, 2015-01-16:
Would it be possible to add an option to future date the resulting transformation from the ekf_localization_node?

I'm trying to use robot_localization along with move_base, but move_base requires that the map->odom transform to be future dated. Global positioning packages like gmapping or AMCL do future date this transformation.

It would be interesting to add the posibility to optionally future date the obtained transformation when using robot_localization as a source for global localization. It would greatly help it's integration with the standar ROS navigation stack.

Thank you and best regards.

ayrton04, 2015-01-19:
Sure, we can do that. Question: do amcl and gmapping future date by default? Also, is it an option for those packages, or does it always future date?

paulbovbel, 2015-01-19:
by default:

amcl future dates transforms by 0.1 seconds
gmapping future dates by the transform_publish_period, which is 0.05 seconds by default. somewhat undocumented.

ayrton04, 2015-03-05:
Moving this to the top of my queue.

ayrton04, 2015-03-05:
Added a `transform_time_offset` parameter which specifies, in seconds, the amount of time to add to the transform's timestamp. For backwards compatibility, it defaults to 0. I still need to update the launch file templates to reflect the change.

closed 2015-03-05 by ayrton04