cra-ros-pkg/robot_localization #158: Parameterizable Queue Sizes

ayrton04, 2015-01-25:
Right now, robot_localization's state estimation nodes assume a queue size of 1. This ensures that the filter always receives the most recent measurements from each sensor. However, it has two significant drawbacks:

1. Users must increase the frequency parameter in order to not miss any measurements from high-frequency sensors.
2. If at time t1 the sensor receives a measurement with timestamp tS1 <= t1, and then at time t2 receives a measurement from another sensor with timestamp tS2 < tS1, then the filter's default behavior is to skip the prediction step and just call correct. There really isn't anything else we can reasonably do in this instance, but we can avoid the situation happening as frequently by decreasing the frequency parameter value and increasing the queue size. Since the filters will order measurements based on timestamps before attempting to integrate them, we'll have fewer cases of old measurements being fused after newer ones. 

This should be exposed as a per-sensor parameter, with a default value of 1 so as to maintain compatibility with previous releases.

ayrton04, 2015-01-25:
This is being handled in the accuracy_fixes branch.

mhkabir, 2015-02-20:
Would it be possible to add a message filter to synchronise the sensor streams? I have synchronised IMU and cameras ( = synced pose estimate) and matching them up makes sense.

ayrton04, 2015-02-26:
This should probably be listed as a new issue, but first I'd like to know more about what you mean and what your use case is. Right now, robot_localization accepts data from multiple asynchronous sources, and I'd prefer not to make the measurements "aware" of one another, but I may be missing your point.

ayrton04, 2015-03-05:
Completed merge of accuracy_fixes branch, which included this change.

closed 2015-03-05 by ayrton04