cra-ros-pkg/robot_localization #163: Reset lastMessageTimes on RosFilter::setPoseCallback

hbadino, 2015-02-24:
Tom,

would you please consider adding the following line to setPoseCallback?

	/// Clear last message times
	lastMessageTimes_.clear();

it happens that I can't really reset the filter by sending a set_pose message and rerun a bag file unless the lastMessageTimes_ map is cleared. Without it, the measurements from the bag file are identified as old and discarded.

Hernan

ayrton04, 2015-02-24:
Yes, I will fix that. Thanks for pointing it out!

hbadino, 2015-02-24:
Tom,

actually, I thought that would solve the problem, but it is not enough. Apparently, calling to set_pose does not make a hard reset of the filter. I see that sometimes, sending the set_pose message works just fine and reset the filter, but some other times it stays in a weird state. There must be some other internal states that must be reinitialized?

Hernan

ayrton04, 2015-02-24:
On what branch are you working?  Check out line 1469 of ros_filter.h in the accuracy_fixes branch. I discovered a little while ago that I wasn't clearing out the measurement queue.

ayrton04, 2015-03-05:
Hi Hernan,

Did my suggestion (clearing out the measurement queue) help?

Thanks,
Tom

hbadino, 2015-03-05:
Tom, 

not really. I don't have the latest version, but I do clear the measurement queue now and filter is still unable to reinitialize correctly. I'll try the latest version sometime in the following weeks to see if it makes a difference.

Hernan

ayrton04, 2015-03-05:
Sounds good. I'll take a deeper look into it. In the meantime, if you get a chance, can you let me know how you're calling the `set_pose` service/topic? Are you doing it via command line, rviz, or programmatically?

hbadino, 2015-03-12:
I'm calling set_pose programmatically. It is going to take me a little longer to check the latest version, but I'll do it eventually.

ayrton04, 2015-04-27:
I think I figured out this issue (or a related one) and have it sorted. I want to test a bit further before I push the changes, but the short answer is that sometimes, after we process the setPose callback, we can get messages with time stamps that predate the setPose message. I now check each sensor message to make certain it's newer than the most recent pose reset.

ayrton04, 2015-04-28:
I think I've addressed this. When you get a chance, give it a try, and let me know if it's working now. If not, feel free to re-open this.

closed 2015-04-28 by ayrton04

hbadino, 2015-04-28:
Thanks Tom. I'll check it.