cra-ros-pkg/robot_localization #166: Add parameter to make estimation wait for all sensors

ayrton04, 2015-03-27:
For some robots, there may be a delay before certain sensor data is available. These delays may result from sensor startup time, launch order, or by design. For example, if a robot starts receiving IMU data immediately but only begins reporting odometry data when it starts moving for the first time, then the estimate error covariance matrix values for the vehicle's position variables will start to explode, as no measurements are arriving for those variables. In this case, what we want is for the filter to only begin estimating once all sensor data is available. It should not, however, stop estimating if said sensor drops out in the middle of a run. 

This feature should be exposed as a per-sensor parameter, and it should default to {{{false}}} for backwards compatibility.

l0g1x, 2015-06-11:
Has this issue been addressed in 2.2.1? Ive actually talked to a few other users, and this issue seems to also effect the initial world_frame-->utm pose if a user has "broadcast_utm_transform" set to true while using navsat_transform_node. If for example ekf_localization is started first, then an imu input is available immediately after, but a gps input is delay by 5+ seconds, the initial world_frame--> utm transform is a value no where close to the initial Datum GPS estimate. I can go into more detail if this seems to be a actual issue, or this possibly may be a user setup error.

ayrton04, 2015-06-11:
In the scenario you describe, everything should work fine, and would be unrelated to this. In fact, the delay between the EKF output and first GPS reception is desired, and is the reason for the "delay" parameter in navsat_transform_node. What value are you expecting to see in the transform vs. what you are seeing? It's the product of your robot's current world frame pose and the *inverse* of the UTM grid position. I'd be interested in seeing a bag or text file output. 

I *think* the only way this issue can affect navsat_transform_node is if you have an IMU and odometry, and you receive some odometry data, then a GPS fix, then your first IMU reading, but that is rare, and can be fixed by the "delay" parameter in navsat_transform_node.