cra-ros-pkg/robot_localization #167: Unexpected behavior when using absolute orientation and angular velocities

jim-rothrock, 2015-04-08:
I have two instances of ekf_localization_node. The first, used for odometry, reads the x velocity (in the base_link frame) from the wheel encoder odometry topic and publishes the odom-to-base_link transform. The second node, used for absolute orientation, reads the x velocity in the same manner as the first node, but also reads the roll, pitch, and yaw angles from the solar compass topic (a PoseWithCovarianceStamped message). The absolute orientation node publishes the map-to-odom transform. Furthermore, the absolute orientation node does not receive poses from the solar compass when the compass is in shadow.

What I am doing:

1. Drive the robot forward until it is in shadow.
2. Make a 90-degree right turn.
3. Drive the robot forward until it is out of shadow.

I am using RViz to display odometry/filtered (published by the odometry ekf_localization node), with the fixed frame set to "map".

What I expect:

1. The odometry display draws arrows that correspond to the robot's path as it moves into shadow.
2. The 90-degree right turn adds no arrows to the odometry path. This is due to the fact that the robot is not translating or, from the absolute orientation node's point of view, rotating. Since the solar compass is in shadow, it does not publish any poses that would be received by the absolute orientation node.
3. When I resume driving the robot  forward, arrows should continue to be added to the odometry path, as if the right turn never took place. When the robot moves out of shadow, arrows added after leaving the shadow should be oriented 90 degrees right, relative to the earlier arrows. The new arrows should form a straight line pointing right.
4. From above, the result should look like this:

    X>>>>>>>>>>
    ^
    ^
    ^
    R->->->->->->
    ^
    ^
    E
    ^
    ^
    ^

E is the point where the robot enters the shadow, R is where the right turn is made, and X is the point where the robot leaves the shadow. The "->" symbols show the actual path of the robot after the right turn.

What actually happens:

    X
    ^
    ^
    ^
    R->->->->->->
    ^
    ^
    E______Y>>>>>>>>>>
    ^
    ^
    ^

The underscores are supposed to be spaces, but I couldn't get non-breaking spaces to work.

When the robot exits the shadow at point X, it is then translated to point Y. It appears that the robot's path between E and X is rotated, instead of just the robot itself. I suspect that the sum of all the position changes caused by x velocity messages since the last pose (orientation) message are being rotated around the position that was current when the last pose message was received. Is this expected behavior? I am using the robot_localization package from the ROS Indigo binary release.

jim-rothrock, 2015-04-09:
I built the indigo-devel version of robot_localization from source, to find out whether the behavior changed. It didn't.

I've attached an RViz screen grab that demonstrates the behavior. The yellow arrows represent the robot's real pose in Gazebo. The red arrows are "odom" frame poses from wheel encoder odometry. The blue and magenta arrows represent odometry/filtered and odometry/filtered_discrete, respectively. The robot is in shadow when it makes the left turn, and the red arrows on the left cease being produced when the robot exits the shadow. At that moment, the robot is translated to the lower right endpoint of the group of red, blue, and magenta arrows on the right.
[image]

ayrton04, 2015-04-09:
Hi Jim,

This looks like a very detailed and well thought-out question! Would you mind re-posting this on the ROS answers site, and also posting a bag and launch file? If the data is proprietary, I understand, but it will make analysis easier. 

Just from reading the description, I have two initial thoughts, both of which may turn out to be completely off the mark once I see the data:

1. Is the unexpected behavior of the robot when it exits the shadow for the orientation-enabled instance of `ekf_localization_node`, or the odometry instance only? If you switch the fixed frame to *odom* and just focus on the odometry-only estimate, does it look as you'd expect? If you then switch the fixed frame back to map and look only at the orientation-and-odometry data, does it look correct? Sometimes viewing the output of one frame from another frame causes non-intuitive visual effects. I'm guessing this isn't your problem, but I just wanted to check.

2. Without seeing data and a launch file, I'm wondering if there isn't an issue with your covariances. When you stop receiving orientation data, the covariance values for your orientation variables will start to explode, and when you then receive new data again, the correlation between orientation and linear velocity can cause jumps in your position. Again, this is only an extremely rough hypothesis, as it's hard to know for certain without seeing a bag file.

Regardless, this looks like a really interesting use case.

jim-rothrock, 2015-04-09:
> Without seeing data and a launch file

Do you mean the launch file that launches the ekf_localization_node instances? Launching my robot makes use of a number of launch files in various directories, so I want to include only what you need.

ayrton04, 2015-04-09:
Yes, just your `ekf_localization_node` launch files will do. If you are broadcasting any transforms that I'll need, then make sure you record the `/tf` topic in the bag file, or, if the transforms are static, just include them. Thanks!

jim-rothrock, 2015-04-10:
I've posted this question at http://answers.ros.org/question/207005/ekf_localization_node-exhibits-unexpected-behavior. I don't want the launch files and bag file to be public, so I've shared my robot_localization Dropbox folder with you.

closed 2015-04-21 by ayrton04