cra-ros-pkg/robot_localization #200: Enhancement Proposal: Ability to conditionally enable/disable sensor data

ben-ebersole, 2015-05-08:
Tldr; I would like the ability to ignore IMU sensor data when the encoders show no motion

I have a mobile robot which is by nature subject to incorrect encoder-odometry data (the navigational base is a clearpath husky, and thus skid-steer), and is unfortunately also subject to an imperfect IMU (slow drift over time). The operations of the robot are such that it alternates between mobile navigation and long periods during which the base is not moving but other operations are taking place. It is during these periods of standstill that the IMU drift is most problematic, as the estimated pose will slowly but continuously rotate over time due to the IMU. However, the robot as a whole has a high enough inertia and operates in a sufficiently docile environment that there is no reason why it should be moving if the wheels themselves are not moving. It would be therefore very helpful to be able to only use the IMU data when the encoder data is showing motion, and ignore it otherwise. Note that removing the IMU entirely from the localization is not an option, as it is essential for actual rotations (due to skid-steer).

I considered downloading and custom modifying the source on my own, but this seems like a practical enhancement from which multiple users may benefit.

ayrton04, 2015-05-08:
Are you using the stock IMU that came with the Husky, or a different model? Does the IMU have a magnetometer that provides an absolute heading reference? I wouldn't expect to see drift unless you had a gyro bias and your IMU didn't have an absolute heading reference. Then again, I think they use imu_filter_madgwick, which may be using the gyro data to generate its final heading estimate. @paulbovbel, can you comment? 

These kinds of enhancements fall into a bit of gray area. In general, I try to assume as little about users' data as possible, especially when it means that I have to monitor the behavior of two different sensors w.r.t. one another, e.g., this is happening with sensor A, and so now I have to treat sensor B differently. Implementing your solution also presents a new problem, which I will lay out in a short vignette:

Time t1: robot is moving along, comes to rest with a heading of 30 degrees.
Time t2: heading starts to drift, but we've recognized that the robot is at rest, and therefore can't be moving, so we either ignore the IMU readings *or* continue to fuse the last one. Robot's heading moves from 30 degrees to 37 degrees while resting.
Time t3: robot starts driving again, but now we have a problem. Which heading do we use? The IMU was reporting a heading of 30 degrees when we stopped. Should we use that or the 37 degree value? Also, if we were ignoring the heading data while at rest, then your heading covariance value will have exploded.

The overhead for keeping track of this kind of thing makes me think it should really be handled by either an intermediary node, or by the IMU driver itself.

If we re-frame the problem, what's really happening is that the IMU's heading error is growing over time. Depending on the source of the issue, you can try a few things. 

1. If the reason you're seeing drift is that your IMU lacks a magnetometer and you're experiencing gyro drift, then when your robot base stops, I would drop the variance on its yaw velocity to something very tiny, say 1e-9, and make sure the yaw velocity value is set to true in your ekf_localization_node config for your odometry data. Just make sure the yaw velocity variance value you use is *much* smaller than the value reported by the IMU when the robot is stationary, and *much* larger when the robot is moving.
2. I think imu_filter_madgwick lets you specify biases for your gyro data. Also, Clearpath has an IMU calibration tool that may help.
3. If I were you, I would throw together a quick node that takes in odometry and IMU data and "freezes" the heading of the IMU data whenever the wheel encoders on the robot read 0. You'll need to keep track of the last heading you had before the robot stopped moving, so that when it starts, you carry on as if that heading is the true heading. You'll basically have a "running offset" from the robot's current heading.

paulbovbel, 2015-05-11:
@mikepurvis for reference

We've had good results using the raw data from the Husky's UM6 with imu_filter_madgwick to generate an filtered IMU estimate for robot_localization.

@ben-ebersole, are you using Husky's standard indigo configuration? How much drift are you accumulating, and in what time-frame?

In the past, Clearpath has experimented with reducing the covariance of odometry during periods of no movement. @ayrton04, would that have the desired effect of overpowering IMU drift? With the switch to diff_drive_controller, this option has disappeared.

ben-ebersole, 2015-05-11:
Thank you for the replies;

I am indeed using the UM6 with imu_filter madgwick, in Husky's standard indigo configuration. I have used the calibration tools provided by Clearpath and while they did improve the situation somewhat they did not eliminate the drift. One hypothesis is that our particular environment (indoors with a good deal of surrounding metal) is disrupting the magnetometer.

@paulbovbel The drift is inconsistent but it is accentuated after turning and then coming to a stop, at which point it has been seen at times to drift ~30 degrees in ~5 seconds

@ayrton04 you bring up a good point, simply ignoring the IMU is not a solution. I'll look into creating that intermediate node to freeze the IMU heading for now.

paulbovbel, 2015-05-11:
@ben-ebersole, the magnetometer will be unreliable indoors at best. 30 degrees in 5 seconds is a lot more than I'd expect however. Please consider opening a ticket at https://support.clearpathrobotics.com/, we can dig into the issue with bags and such, to try and determine if this may be a hardware failure.

If you delete the calibration file in /etc/ros/husky_bringup, imu_filter_madgwick should use the default config file which disables the magnetometer (https://github.com/husky/husky_robot/blob/indigo-devel/husky_bringup/config/mag_config_default.yaml). Does this make your situation better or worse?

ayrton04, 2015-05-11:
@paulbovbel, it really depends on the source of the drift. IIRC the Husky's IMU config settings for ekf_localization_node include the absolute yaw data from the IMU's mag, so it probably won't help that much. If the problem was a result of the EKF integrating a biased gyro (and you had no absolute reference), then dropping the yaw velocity covariance in the odometry data would be beneficial. It sounds like the absolute yaw coming out of the IMU filter is drifting, though. 

Regardless, I'll close this issue, but feel free to continue the conversation here.

closed 2015-05-11 by ayrton04

ben-ebersole, 2015-06-08:
I just realized I left this issue somewhat unresolved..

@paulbovbel, I didn't end up opening a ticket because we discovered that our issue was with AMCL, not the imu itself. When i added the intermediate node discussed earlier I successfully eliminated the IMU drift, but our localization in RVIZ was still showing large drifts, particularly after the robot came to a stop after a rapid motion. It appears to be a lag issue in the AMCL computations that we're still working on. 

The IMU does still have some drift due to the issues with the magnetometer, but it is much less than what I'd reported earlier, which was due to AMCL.

Thanks for the help!

paulbovbel, 2015-06-08:
@ben-ebersole, thanks for the follow up, glad things worked out