cra-ros-pkg/robot_localization #201: missing header file "robot_localization/SetDatum.h"

Zachwiany, 2015-05-08:
In the header file [navsat_transform.h](https://github.com/cra-ros-pkg/robot_localization/blob/master/include/robot_localization/navsat_transform.h) there is a library, which is missing.
I mean the line.
```ruby
33 #include "robot_localization/SetDatum.h".
```
Anyone know how can i get or where to find it?

ayrton04, 2015-05-08:
You have the latest source? That's a header that should be getting auto-generated from the SetDatum service message. Which branch are you using?

Zachwiany, 2015-05-08:
I downloaded it a few hours ago so it's probably the latest one, from the master branch, as I though it will be created during package will be make. Currently i don't have acces to robot and his computer for a few days, but will check it again next week, maybe I just did something wrong. 
Thank you for response.

ayrton04, 2015-05-08:
No problem. I will pull down the master branch and try to build it from scratch to see if I can replicate it. I thought I had the build flags and dependencies set correctly, but I could be wrong.

ayrton04, 2015-05-15:
Have you looked into this any more yet? I sourced `/opt/ros/indigo/setup.bash` to reset my ROS config, and then cloned this source, checkout out the master branch, and ran `catkin_make`. Here's the output:

```
-- Generating done
-- Build files have been written to: /Users/ayrton/ros_staging/build
####
#### Running command: "make -j4 -l4" in "/Users/ayrton/ros_staging/build"
####
Scanning dependencies of target std_msgs_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_py
Scanning dependencies of target std_msgs_generate_messages_cpp
Scanning dependencies of target _uuid_msgs_generate_messages_check_deps_UniqueID
[  0%] [  0%] [  0%] Built target std_msgs_generate_messages_py
Built target std_msgs_generate_messages_cpp
Built target std_msgs_generate_messages_lisp
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GetGeographicMap
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_RouteSegment
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_MapFeature
[  0%] [  0%] Built target _uuid_msgs_generate_messages_check_deps_UniqueID
Built target _geographic_msgs_generate_messages_check_deps_MapFeature
[  0%] [  0%] Built target _geographic_msgs_generate_messages_check_deps_GetGeographicMap
Built target _geographic_msgs_generate_messages_check_deps_RouteSegment
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_BoundingBox
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GeographicMapChanges
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GetRoutePlan
Scanning dependencies of target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GeoPose
[  0%] Built target _geographic_msgs_generate_messages_check_deps_GeographicMapChanges
[  0%] Built target _geographic_msgs_generate_messages_check_deps_BoundingBox
[  0%] Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_KeyValue
Built target _geographic_msgs_generate_messages_check_deps_GetRoutePlan
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_RouteNetwork
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GeoPoint
[  0%] Built target _geographic_msgs_generate_messages_check_deps_GeoPose
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_GeographicMap
[  0%] [  0%] Built target _geographic_msgs_generate_messages_check_deps_KeyValue
Built target _geographic_msgs_generate_messages_check_deps_GeoPoint
[  0%] Built target _geographic_msgs_generate_messages_check_deps_RouteNetwork
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_RoutePath
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_UpdateGeographicMap
Scanning dependencies of target _geographic_msgs_generate_messages_check_deps_WayPoint
[  0%] Built target _geographic_msgs_generate_messages_check_deps_GeographicMap
Scanning dependencies of target geometry_msgs_generate_messages_cpp
[  0%] Built target _geographic_msgs_generate_messages_check_deps_UpdateGeographicMap
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _geographic_msgs_generate_messages_check_deps_WayPoint
Scanning dependencies of target geometry_msgs_generate_messages_lisp
Scanning dependencies of target filter_utilities
[  0%] Built target _geographic_msgs_generate_messages_check_deps_RoutePath
Scanning dependencies of target _robot_localization_generate_messages_check_deps_SetPose
[  0%] Built target geometry_msgs_generate_messages_lisp
Scanning dependencies of target _robot_localization_generate_messages_check_deps_SetDatum
Scanning dependencies of target ros_filter_utilities
[  0%] Built target _robot_localization_generate_messages_check_deps_SetPose
[  0%] Built target _robot_localization_generate_messages_check_deps_SetDatum
Scanning dependencies of target sensor_msgs_generate_messages_lisp
[  0%] Scanning dependencies of target rosgraph_msgs_generate_messages_py
Built target sensor_msgs_generate_messages_lisp
[  0%] Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_py
[  0%] Scanning dependencies of target roscpp_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_lisp
[  0%] Scanning dependencies of target actionlib_generate_messages_cpp
Built target roscpp_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_cpp
Scanning dependencies of target actionlib_generate_messages_py
[  1%] [  1%] Scanning dependencies of target roscpp_generate_messages_py
Built target actionlib_generate_messages_py
Building CXX object robot_localization/CMakeFiles/filter_utilities.dir/src/filter_utilities.cpp.o
[... 127 lines cut ...]
Linking CXX shared library /Users/ayrton/ros_staging/devel/lib/libgeoconv.dylib
Linking CXX shared library /Users/ayrton/ros_staging/devel/lib/libekf.dylib
[ 93%] Built target geoconv
[ 93%] Built target ekf
Linking CXX shared library /Users/ayrton/ros_staging/devel/lib/libnavsat_transform.dylib
[ 93%] Built target navsat_transform
Scanning dependencies of target navsat_transform_node
[ 95%] Building CXX object robot_localization/CMakeFiles/navsat_transform_node.dir/src/navsat_transform_node.cpp.o
[... cut ...]
[ 98%] [100%] Building CXX object robot_localization/CMakeFiles/ukf_localization_node.dir/src/ukf_localization_node.cpp.o
Building CXX object robot_localization/CMakeFiles/ekf_localization_node.dir/src/ekf_localization_node.cpp.o
Linking CXX executable /Users/ayrton/ros_staging/devel/lib/robot_localization/ekf_localization_node
Linking CXX executable /Users/ayrton/ros_staging/devel/lib/robot_localization/ukf_localization_node
[100%] Built target ekf_localization_node
[100%] Built target ukf_localization_node
```

Zachwiany, 2015-05-15:
Hello, today i was able to built it, but didnt have enough time to test it. I have no idea what I omitted the last time.
Thank you very much for your help.

ayrton04, 2015-05-16:
No worries! Just re-open this issue if it recurs.

closed 2015-05-16 by ayrton04