cra-ros-pkg/robot_localization #204: Build error with tf2 changes

mryellow, 2015-05-19:
```
navsat_transform.cpp:113:50: error: no matching function for call to ‘fromMsg(const _pose_type&, tf2::Transform&)’
     tf2::fromMsg(msg->pose.pose, latestWorldPose_);
```

`latestWorldPose_` is intended to be a pose or transform? Someone halfway through working on this part?

ayrton04, 2015-05-19:
I think you need to update your `tf2` libraries. Look at `tf2_geometry_msgs/tf2_geometry_msgs.h`. If that method is missing, you need to update. Try `sudo apt-get update` and `sudo apt-get upgrade`, and try again. 

Let me know how it goes!

mryellow, 2015-05-19:
I looked at that so many times and didn't find the signature, then I punch it in google this-morning and this is first result:

http://docs.ros.org/indigo/api/tf2_geometry_msgs/html/tf2__geometry__msgs_8h.html
`tf2::fromMsg (const geometry_msgs::TransformStamped &msg, geometry_msgs::TransformStamped &out)`

Must have been looking at an older version of the API docs which didn't show the `(pose, transform)` signature.

Thanks for the heads-up, got a bit of a funky source install so things like that can happen occasionally. Will close now and comment again when I get a result.

closed 2015-05-19 by mryellow

mryellow, 2015-05-19:
The packages in `geometry_experimental` were specified in my `.rosinstall` at  `0.5.7-0` rather than `0.5.11-0`. 

Something I haven't been caught up by before, interesting that catkin doesn't specify version numbers in `package.xml`, with webdev we have package managers Bower, NPM and the like which use `~0.5.7` style definitions to grab the most recent compatible code.

`[build] Finished.` Thanks again.