cra-ros-pkg/robot_localization #214: version 2.2.1 tf2 issues

PatrickHussey, 2015-06-02:
Version 2.2.1 complains about the '/' prefix for frame ids and will not run. 2.1.7 works fine.

rqt_console outputs these warnings continuously:

Could not obtain transform from /segway/base_link to segway/base_link. Error was Invalid argument "/segway/base_link" passed to lookupTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 

I understand that in tf2 whatever string you pass in is your frame_id. So "my_frame" is simply "my_frame" and "/my_frame" would be a different frame_id. To avoid silent failures tf2 does not accept frame_ids starting with "/" with an tf2::InvalidArgument exception.

Not sure you need to "fix" anything since it adheres to the standard for tf2; rather maybe just something on the wiki explaining or more verbose errors in the code

ayrton04, 2015-06-02:
Agreed, the wiki needs a lot of attention. I think part of the warning is actually from within the tf2 library itself. I just pass on the exception error message. I'll make sure to add this to the wiki when I update it, and will close the issue when it's done. Thanks!

ayrton04, 2015-06-07:
I added a small note in the tutorials:

http://wiki.ros.org/robot_localization/Tutorials/Preparing%20Your%20Sensor%20Data#Handling_tf_prefix

Feel free to edit the wiki page if you feel this could use clarification!

closed 2015-06-07 by ayrton04

Jasmin92, 2016-11-18:
Hi! I'm trying use the package in my robot simulation, but I'm getting this warning :
[ WARN] : Could not obtain transform from /world to base_footprint. Error was Invalid argument "/world" passed to lookupTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 
the gps data  generated by gazebo has "/world" as frame id . So the navsat_transform is not working and the GPS is not included in the pose estimation. Is there any way to avoid the issue. I was thinking of changing the GPS data frame id but I don't know how to change it in gazebo. I know that the warning is from the tf2 lib and this may be the wrong place to ask  but your help would be appreciated.

ayrton04, 2016-11-22:
See the [definition of the NavSatFix message](http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html). To quote it:

> header.frame_id is the frame of reference reported by the satellite receiver, usually the location of the antenna.  This is a Euclidean frame relative to the vehicle, not a reference ellipsoid.

In other words, the `frame_id` in a NavSatFix message should be the location of the sensor on the vehicle itself, i.e., its offset from _base_link_. You then need to define a transform from _base_link_ (or whatever your robot's body frame is) to the GPS frame. In this case, it should go from _base_link_->_world_. If you can, I'd see if there's a parameter in that plugin to rename the GPS `frame_id` to something like _gps_ and then define a _base_link_->_gps_ transform that provide the offset position of the GPS sensor on your simulated robot.

The error you're seeing is telling you that it needs that transform, and you haven't defined it.

ItsJas, 2016-11-22:
Thanks for your answer Tom.
 I fixed the problem, the gps frame_id's default value is normally "" I don't know what is setting it to /world. Anyway I forced it to be gps by adding <frameId>gps</frameId> in the gps plugin's integration in my robot's urdf. The transform was already defined as base_link -> gps_frame_id, the warning was basically caused by the '/' & tf2.