cra-ros-pkg/robot_localization #223: Fix gravitational acceleration correction when orientation data is missing

ayrton04, 2015-07-30:
If the orientation data isn't filled out in an IMU message (per its [specification](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html)) and the `remove_gravitational_acceleration` parameter is set to *true*, then when we attempt to remove the gravitational acceleration, we get nans in the measurement (which are ignored, but it shouldn't get that far). We should print a diagnostic message in this case and either ignore the measurement or fuse it without correcting it. I think I prefer the former.

kphil, 2016-01-15:
An IMU does not provide orientation by default unless there is an attitude computer (and most commonly a magnetometer). Therefore it would be nice being able to use an IMU just providing acceleration and angular velocities with `remove_gravitational_acceleration` parameter set to true.
I would suggest to use orientation from our filters state vector to remove gravity in this case.

ayrton04, 2016-01-15:
The rotation angles that matter for correcting linear acceleration are roll and pitch, which are typically provided by accelerometer data, yes? You may be able to pass the accelerometer data through another node like imu_filter_madwick as well, which would give you the orientation values you need. In any case, yes, I agree that this should be a feature, but see my comment on the change you proposed.

kphil, 2016-01-17:
Yes, there are many ways to provide an orientation and many IMUs come with an built-in attitude computer. But if i have a minimal setup with an IMU just providing acceleration and angular velocity it would be nice being able to run it with this package. That's the idea, if someone will provide an orientation we will use it, if not using current states orientation.

ayrton04, 2016-01-17:
Sounds good. Thanks for the PR!

closed 2016-01-17 by ayrton04