cra-ros-pkg/robot_localization #225: navsat_transform_node does not publish GPS odometry when using manual datum

v01d, 2015-08-07:
I'm trying to run navsat_transform and I don't see any /odometry/gps published.

I only see this output:
[ INFO] [1438961738.845787829]: Datum (latitude, longitude, altitude) is (49.275700, -122.896000, 0.000000)
[ INFO] [1438961738.845886373]: Datum UTM coordinate is (507564.748074, 5458109.981001)
[ INFO] [1438961738.845966495]: Initial odometry position is (0.000000, 0.000000, 0.000000)

It seems that something is missing since I know it should print more things.
The roslaunch code I'm using is:

      <node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true" output="screen">
        <param name="yaw_offset" value="1.5707963267948966"/>
        <param name="zero_altitude" value="true"/>
        <param name="publish_filtered_gps" value="false"/>
        <param name="broadcast_utm_transform" value="true"/>
        <param name="use_odometry_yaw" value="false"/>
  
        <param name="wait_for_datum" value="true"/>
        <rosparam param="datum">[49.2757225, -122.895907, 134.4, map, base_link]</rosparam>
    
        
        <remap from="/gps/fix" to="/navsat/fix"/>
        <remap from="/imu/data" to="/robot/imu"/>
        <remap from="/odometry/filtered" to="/robot/pose"/>
      </node>

When  wait_for_datum is false, I get:

[ INFO] [1438962039.036361347, 1429484457.940453985]: Initial odometry position is (76.104844, -15.656690, 0.000000)
[ WARN] [1438962039.141576283, 1429484458.041111175]: Could not obtain transform from imu_link to base_footprint. Error was Lookup would require extrapolation into the past.  Requested time 1429484458.044268163 but the earliest data is at time 1429484458.335172013, when looking up transform from frame [imu_link] to frame [base_footprint]

[ INFO] [1438962039.141708366, 1429484458.041111175]: Datum (latitude, longitude, altitude) is (49.275600, -122.895915, 136.500000)
[ INFO] [1438962039.141750354, 1429484458.041111175]: Datum UTM coordinate is (507570.933931, 5458098.909493)
[ INFO] [1438962039.141795890, 1429484458.041111175]: Initial odometry position is (76.202513, -15.705080, 0.000000)
[ INFO] [1438962039.141845494, 1429484458.041111175]: Corrected for magnetic declination of 0.000000 and user-specified offset of 1.570796. Transform heading factor is now -0.956721
[ INFO] [1438962039.141896643, 1429484458.041111175]: Transform world frame pose is: 
Position: (76.202513, -15.705080, 0.000000)
Orientation: (0.000000, 0.000000, -0.460000)
[ INFO] [1438962039.141948952, 1429484458.041111175]: World frame->utm transform is 
Position: (2154878.326476, -5040382.731929, -136.500000)
Orientation: (0.000000, -0.000000, 0.496721)

ayrton04, 2015-08-10:
Are you building from source, or did you use the package manager?

v01d, 2015-08-10:
Forgot to update this. I built from source and it started working. I guess it was an old version.

closed 2015-08-10 by v01d

ayrton04, 2015-08-10:
Nah, not old necessarily. The issue is that the datum handling wasn't quite done correctly in the released version, which is why I don't advertise those parameters on the wiki. I've since fixed it. Sorry about that!

v01d, 2015-08-10:
Ok, thanks for info!