cra-ros-pkg/robot_localization #232: Add support for Range message

mhkabir, 2015-08-29:
http://docs.ros.org/jade/api/sensor_msgs/html/msg/Range.html
Would be glad to send in a PR with the support for the Range message as a ground-distance sensor, if you can point out the relevant parts where I'd need to make changes :)

ayrton04, 2015-08-29:
I don't think I'll want to integrate this with `ekf_localization_node` or `ukf_localization_node` directly, as those nodes only accept data that is directly measuring a state variable, and I'd like to keep it that way. There's really no specialized logic to convert raw sensor data into measurements that can be fused by the filters; almost all pre-processing of the input messages involves coordinate frame transformations.

What you are proposing should be a node in itself that takes in whatever data it needs, along with the range message, and outputs a PoseWithCovarianceStamped message that can be fused with one of the filters. It's pretty much directly analogous to `mavsat_transform_node`, which takes GPS, IMU, and odometry information and produces a message that the filters can consume. However, others may have use for such a node, so I'd be interested in accepting a PR for a brand new node that can be included in the package. If you're interested, we should solidify exactly what the node would be doing so as to make it as broadly applicable as possible.

mhkabir, 2015-09-01:
@ayrton04 Sounds good, I'm all game. A new node would be perfect.

closed 2015-12-21 by mhkabir

quentin1002, 2016-03-03:
Hi @mhkabir 

Did you manage to create the node you were talking about? 
I have troubles interfacing the px4 + px4flow with robot_localization via mavros. The only ready to use topic coming out of mavros (in a GPS-less situation) is the IMU... Getting the ground distance into robot_localization would be great!! 
Thanks