cra-ros-pkg/robot_localization #234: IMU data ambuiguity

mhkabir, 2015-09-06:
The ROS IMU message spec is rather weird, since it contains fields for the *estimated* attitude (orientation), which comes from an AHRS in some form.
The other two fields, acceleration and angular velocities are actually raw data fields from the accelerometer and gyroscope. My question is, what does robot_localization expect, raw or fused data? Because mixing both seems weird.
Also, the message spec is such that magnetometer measurements have no space.

How do we resolve this ambiguity? The discussion on the new message spec which should've resolved this seems to have dried up : https://groups.google.com/forum/#!topic/ros-sig-drivers/Fb4cxdRqjlU

ayrton04, 2015-09-06:
Looking at the [spec](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html), I don't see any declaration of what is assumed (raw or filtered) in each field. Can you point me to your source for that?

In any case, robot_localization makes no assumption. If your IMU reports previously filtered data, then it will just result in a slightly longer delay between the time your IMU moves and the time that motion is reported. We'll see how the message spec shakes out, but in general, the only assumption r_l makes is that by the time the data reaches the EKF or UKF, it's ready to be fused with the other sensor data.

mhkabir, 2015-09-06:
@ayrton04 
Re-iterating on that, in line with your observations, I actually see that nothing is mentioned for raw/filtered. But by definition, the orientation has to come from an AHRS (pre-filtered) while the linear_acceleration and angular_velocities could be either. Looking at various IMU drivers, I see that many mix raw/filtered data in the message, while others fill accel/velocities with AHRS-corrected values.

I'm planning to use robot_localisation in a high-performance flight loop, and I guess that using raw data makes the most sense for me.

But then, this approach has atleast 2 issues that I can see : 
* I have no way of filling the orientation field in the IMU message. How do I handle this in r_l ?
* No global magnetometer data can be fused due to the lack thereof in the message spec. This would yield yaw un-observable, and a large integration drift would be there in the attitude estimate (making it practically useless for me)
* This goes against r_l's framework to only fuse data which directly modifies the system state, and quoting you : 
> There's really no specialized logic in those nodes to convert raw sensor data into measurements that can be fused by the filters

ayrton04, 2015-09-06:
This would be a good question for the ROS answers site.

First bullet point: set the orientation variables to false for your IMU configuration
Second bullet point: are you saying you have a magnetometer, but because it's not in the IMU message spec, you can't use it? I suggest you look at [this](http://wiki.ros.org/imu_compass) or [this](http://wiki.ros.org/imu_filter_madgwick).
Third bullet point: All I meant is that your sensor needs to provide one of the following: 
X position
Y position
Z position
X velocity
Y velocity 
Z velocity
Roll
Pitch
Yaw
Roll Velocity 
Pitch Velocity
Yaw Velocity
X linear acceleration
Y linear acceleration
Z linear acceleration

If your sensor/data source provides one of these, it will work fine with the EKF or UKF. On the other hand, if you try to give it raw magnetometer data, then it needs to do some specialized logic to turn those measurements into one of the state variables, and then fuse it with the state estimate. If you give it range measurements, it would have to use specialized logic to turn those into position measurements. Any logic that is required to turn the data in question into one of the state variables above is considered outside the scope of the node. The more of the specialized logic that you push into the filter nodes themselves, the less generic they become, and users have to work harder to integrate them with their systems (that, or the code becomes unnecessarily bloated). The nice part of ROS is that you can use whatever node you want to covert from mag data to heading, and just throw that in between your IMU and the EKF.

closed 2015-09-13 by ayrton04