cra-ros-pkg/robot_localization #235: Feature request : Using Quaternion representation instead of Eulers.

mhkabir, 2015-09-06:
I had a look at the code, and I see that the filter internally uses Euler angles in it's state representation. In aerospace systems, this can lead to gimbal lock (https://en.wikipedia.org/wiki/Gimbal_lock) in various situations, and I was hoping that we could fix this.

ayrton04, 2015-09-06:
Yes, I'm well aware of this, and there are people investigating this now. It would constitute massive changes throughout the software, and I may not get the cycles to address it any time soon. The best case would be to create new nodes that use quaternions and keep them separate from the existing filters. Once they're working well, we can replace the existing ones.

ayrton04, 2016-08-01:
Update: I'm working on this as time allows. I have put together the transfer function and Jacobian for the omnidirectional motion model on paper, and just need to convert it to code.

brawner, 2021-08-05:
Do you have any updates on this feature? Our application cares strongly about orientation and it would be great to use this solution.

If you're able to pose the math you have already worked out, I may be able to also help with the implementation.

SteveMacenski, 2021-08-05:
Given it hasn't been touched since 2017, I don't think there are any updates.

ayrton04, 2021-08-05:
I don't see this feature happening, frankly. I recommend taking a look at [fuse](https://github.com/locusrobotics/fuse). It may require some additional models to be written, but the scope of this change is really too large to be taken on in a package that is really only being maintained at this point.

brawner, 2021-08-05:
Thanks for pointer @ayrton04, will take a look