cra-ros-pkg/robot_localization #245: Could not find a package configuration file provided by "geographic_msgs"

772358053, 2015-12-03:
When I build the robot_localization package from source, I got a CMake Error like this:
```
-- ==> add_subdirectory(robot_localization)
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  robot_localization/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/siat/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/siat/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] 错误 1
Invoking "make cmake_check_build_system" failed

```
Could anyone help me?

ayrton04, 2015-12-03:
`sudo apt-get install ros-indigo-geographic-msgs`

You can also try `rosdep install robot_localization`

closed 2015-12-03 by ayrton04

772358053, 2015-12-03:
@ayrton04 It's great! The command  ```sudo apt-get install ros-indigo-geographic-msgs``` solve the problem. Thank you very much!

kumarrt, 2018-05-01:
I was getting the same error, I tried `sudo apt-get install ros-kinetic-geographic-msgs` and it worked.

ayrton04, 2018-05-02:
If you have a catkin workspace and are building any code from source, your first step should always be

```
cd /your/catkin/workspace/root
rosdep install --from-paths src --ignore-src
```

This will automatically install all required binary dependencies for the packages in your workspace.

VictorBLop, 2020-06-02:
Thank you very much, the first idea solved the problem for me.

zeelbhatt, 2021-04-24:
I installed all dependencies using `rosdep install --from-paths src --ignore-src` 

but still have same problem

cl12312, 2024-11-01:
1