cra-ros-pkg/robot_localization #248: Sensor rate vs. filter rate vs. sensor queue?

kogut, 2016-01-15:
Ubuntu 14.04 / Indigo

I was having some problems with a sensor configuration.  In trying to isolate the problem I pared my configuration down to a single 1000Hz rate gyro (using bagged data).

I am adding only one field (yaw twist) to the filter via an imu message.

I've noticed this (same with ukf/ekf):

1) frequency 10: no immediately noticeable problems.  Robot turns in place according to real-life turns, smoothly 

2) frequency 100:  the position output is jumpy.  Though the frequency is indeed 100, the yaw position of the robot will "lock" in place over multiple samples...a few tenths of a second at times.   And lock in place with the *exact* same numerical numbers for yaw and yaw covariance.  Sometimes the filter will lock up entirely.  It doesn't crash or complain (checked rosout and the output of the debug file, etc).  And the sensor data rolls on.  It just stops publishing and its CPU usage drops a lot.  

3) frequency 1000:  2), but exacerbated.   The filter sometimes never outputs data at all.  Sometimes it'll output for a tenth of a second or so.  

In all the above the output data looks reasonable (when it's being output).  The covariance isn't exploded or anything, and the yaw looks reasonable.   It's probably wrong vs. ground truth, but the robot isn't spinning wildly or anything. 

Focusing on 2), if I increase the "imu0_queue_size" to 100, the problem is temporarily held at bay (or masked)  The output is smooth.  But at then, a short period later, say 5-10 seconds, the filter abruptly stops publishing.  Again, no errors/warnings/suspicious data.   If I set it to a very large value, e.g. 1000+, then everything is smooth over the course of my bag (but I haven't checked for accuracy vs. truth).

It appears that the queue_size is a backdoor sensor rate decimation?

Is this expected behavior?  Are there known parameter knobs to turn to correct it?   (didn't find much in ROS answers or the documentation). 

I have a relatively fast vehicle, so I have a preference for having high-rate ekf output and full-use of high-rate data vs. just decimating the sensor data or dialling back the frequency to <100Hz.

ayrton04, 2016-01-15:
Do you have sample data you can provide? I've seen this run at speeds much higher than 100 (though I'd be suprised if you reach 1000). So far I've only seen frozen/locked filters when the covariance input has an issue and we end up with nans somewhere.

kogut, 2016-01-15:
I'll prepare a bag.  First I'm doing one more experiment with synthetic 1000Hz data to see if I can replicate the issue.   I'd be surprised if the covariance input is an issue since the sensor outputs a static covariance value that's reasonable.  It's a low value, 0.001, but the sensor itself is a fiber-optic gyro that is indeed very low noise.

ayrton04, 2016-01-15:
What variables are you measuring in your r_l config? Maybe post it here?

kogut, 2016-01-15:
<node pkg="robot_localization" type="ekf_localization_node" name="robot_localization_odom" output="screen" clear_params="true">
            <param name="frequency" value="100"/>
            <param name="sensor_timeout" value="0.4"/>
            <param name="two_d_mode" value="false"/>
            <param name="map_frame" value="$(arg map_frame)"/>
            <param name="odom_frame" value="$(arg odom_frame)"/>
            <param name="base_link_frame" value="$(arg base_link_frame)"/>
            <param name="world_frame" value="$(arg odom_frame)"/>
            <param name="transform_time_offset" value="0.0"/>

            <param name="imu0" value="/localization/sensors/imu/kvh/raw"/>
            <rosparam param="imu0_config">[false, false, false,
                                                false, false, false,
                                                false, false, false,
                                                false, false, true,
                                                false, false, false]</rosparam>
            <param name="imu0_differential" value="false"/>
            <param name="imu0_relative" value="false"/>
            <param name="imu0_remove_gravitational_acceleration" value="false"/>
            <param name="imu0_queue_size" value="100"/>

           <param name="print_diagnostics" value="false"/>
           <param name="debug"           value="false"/>
           <param name="debug_out_file"  value="debug_ekf_localization.txt"/>

  
 
</node>

ayrton04, 2016-01-15:
Sorry, can you paste the whole thing? I want to see what variables you're measuring from all sensors.

kogut, 2016-01-15:
Yeah, sorry, part of its not showing up probably because of some unseen character, and I'm trying to edit it now...bear with me for a minute while I sort it out.

kogut, 2016-01-15:
<rosparam param="process_noise_covariance">[0.05, 0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0.05, 0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0.06, 0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0.03, 0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0.03, 0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0.06, 0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0.025, 0,     0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0.025, 0,    0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0.04, 0,    0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0.01, 0,    0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0.01, 0,    0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0.02, 0,    0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0.01, 0,    0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0.01, 0,
                                                            0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0.015]</rosparam>
            <rosparam param="initial_estimate_covariance">[1e-9, 0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    1e-9, 0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    1e-9, 0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    1e-9, 0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    1e-9, 0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    1e-9, 0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    1e-9, 0,    0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    1e-9, 0,    0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    1e-9, 0,     0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    1e-9,  0,     0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     1e-9,  0,     0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     1e-9,  0,    0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     1e-9, 0,    0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    1e-9, 0,
                                                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    1e-9]</rosparam>

kogut, 2016-01-15:
[localization.launch.txt](https://github.com/cra-ros-pkg/robot_localization/files/92586/localization.launch.txt)

kogut, 2016-01-15:
[localization.launch.txt](https://github.com/cra-ros-pkg/robot_localization/files/92588/localization.launch.txt)

kogut, 2016-01-15:
There you go.  The topics with "_no" are ones I effectively removed as filter inputs for debugging purposes.

kogut, 2016-01-15:
I should add that imu1 outputs a full complement of IMU data, what's shown is my testing it as purely a lower-rate yaw-rate sensor to compare with the higher-rate gyro.   And I found that the issues don't seem to exist for the lower-rate sensor, though I haven't done a full range of comparative tests.

ayrton04, 2016-01-15:
Yeah, I'd have to see a bag. As the launch file is now, I would expect your output covariance to get very ugly very quickly, because you're only measuring one variable with an omnidirectional motion model.

ayrton04, 2016-01-15:
Oh, and if you do a bag, don't run the EKF. I just want raw sensor data, please.

ayrton04, 2016-01-15:
(...and transforms)

kogut, 2016-01-15:
Roger, on data and tf.  That's going to take me a little bit because the sensor tfs are coming from a complex URDF, and I'd prefer to give you a bag with just the one sensor and with the sensor frame as base_link so I don't have to mess around with recording a bag with a ton of unnecessary tf data.

kogut, 2016-01-15:
Understand about the covariance, but I was seeing this with a full complement of other sensor data...thought I can't say for certain that this is the real issue....certainly possible that I went down the wrong rabbit hole while debugging.

kogut, 2016-01-15:
Just for reference this is my "desired" config, which has the same freezing issue at the same time, but really only with the KVH topic active.   

I'll work on that bag now.  :)   

[localization.launch.txt](https://github.com/cra-ros-pkg/robot_localization/files/92634/localization.launch.txt)

kogut, 2016-01-16:
Preparing the bag led me to isolate the problem.  It's a tf lookup problem.   When I tell r_l to use the sensor frame as the base_link frame, all issues disappear and r_l runs beautifully at any rate.  Even with ill-conceived r_l configurations.   When I'm running my bags my tfs are coming live from robot_state_publisher/URDF with use_sim_time set to true.    With the robot_state_publisher publishing at a high enough frequency that r_l should have no problem.  My running theory is that some characteristic of simulated time is confusing/locking the tf lookup.   I'll verify this, then close this issue.  Thanks for the help, and sorry for the waste of time.

kogut, 2016-01-16:
OK, the issue is that my robot_state_publisher did not have the param "use_tf_static" set to true, which is the default in Jade, but not in Indigo.   And high-frequency non-static tfs don't appear to work very well with r_l.  Simply explicitly turning on "use_tf_static" fixed the issue.  Thanks for the help.

closed 2016-01-16 by kogut

ayrton04, 2016-01-17:
Interesting! Thanks for the information. It's good to hear these kinds of things so others can benefit. In fact, it might make a good question to ask on ROS Answers, even if you just answer it yourself. Does robot_state_publisher use tf or tf2? If it's still tf, then even static transforms are broadcast repeatedly, as opposed to tf2, where they are latched. r_l uses tf2 for that reason. There may be some trouble with the use of the tf2 MessageFilters, but I'd have to investigate further.

kogut, 2016-01-19:
Yeah, robot_state_publisher has been ported to tf2, so the use of the use_tf_static param definitely reduces tf SPAM traffic, latency in lookups, and should definitely be a "best practice" when used with r_l.  Fortunately it's the default now in Jade...I'm just still on Indigo.