cra-ros-pkg/robot_localization #251: Robot 3D kinematic model

shisha101, 2016-01-22:
I would like to have a look at the 3D kinematic model being used for estimation, I have looked through the paper but was not able to find any reference to where the model came from.
Could someone point me as to where the model came from and where I can have a look at it.

ayrton04, 2016-01-22:
http://answers.ros.org/question/221837/robot_localization-ekf-internal-motion-model/#221944

shisha101, 2016-01-24:
Thanks Tom!

I want to implement and hopefully contribute another model specifically for differential drive robots the type corresponding to the paper would be (2,0). I believe this should be more accurate with respect to the prediction step and would be quiet useful to have, as many mobile robots use differential drive. I want to localize a differential drive robot outdoors using gps(rtk), Imu and odometry.

Could you point me to the material that was used to generalize the 2D models in the paper to 3D, and to any material you might deem useful?

ayrton04, 2016-02-01:
Velocities are given in the body frame of the robot, so I just needed to perform a 3D rotation of the velocities. It's really just a matter (for Euler angles) of multiplying three individual rotation matrices:

http://inside.mines.edu/fs_home/gmurray/ArbitraryAxisRotation/

I should really be using quaternions, but then that's already on my to-do list. :)

shisha101, 2016-02-03:
Thanks again !

closed 2016-02-03 by shisha101

diasdm, 2018-06-20:
Hello,
Sorry for posting in an old issue but my questions are closely related with this one.
* To my understanding this was never implemented, am I right?
* Is this worth it? (To my understanding the way it has been done so far is by fusing the value 0, with a low covariance, to the y axis velocity, as mentioned [here](http://docs.ros.org/lunar/api/robot_localization/html/configuring_robot_localization.html), and apparently it has been working well.)
* Is this something that you hoped it would be implemented but never was?

Thank you for your time.

ayrton04, 2018-06-20:
We have only one kinematic model in the state estimation nodes, correct. It is omnidirectional. If you feed it the right values, it effectively becomes the unicycle model in 2D.

diasdm, 2018-06-20:
Just to be clear, for a differential drive robot you don't think there would be an improvement in the estimation error if the kinematic model was the unicycle instead of the omnidirectional? Considering that both were feed with the right values.

ayrton04, 2018-06-20:
The last time I investigated this, I believe I found that the kinematic model that we are using is equivalent to the unicycle model when you feed 0 values to all the 3D variables & Y. While I agree that a strict 2D kinematic model would be great, (a) the CPU usage for the node is already very low, (b) the model we're using is intentionally generic so as to support a wide range of robots, and (c) I don't have the cycles to add a new model. I'd welcome a PR on that front, though I'd [caution you about investing too much time in it either](https://github.com/cra-ros-pkg/robot_localization/issues/265#issuecomment-396721380).

diasdm, 2018-06-21:
Thank you for your view on the subject and for the additional information. Looking forward to hear more about the (possible) plans for the new package! In the meantime I'll take a look at the code and see what I can do without spending too much time.