cra-ros-pkg/robot_localization #253: Warning about transformation

akifh, 2016-02-02:
Hi all. I'm running ekf_localization with odom and imu. In console it outputs a warning on transformation about once every 1-2 secs;

```
Node: /ekf_localization
Time: 12:14:21.414057836 (2016-02-02)
Severity: Warn
Published Topics: /diagnostics, /odometry/filtered, /rosout, /tf

WARNING: failed to transform from base_imu_link->base_footprint for imu0 message received at 1454408061.356009006. No transform exists from source to target frame.

Location:
/tmp/buildd/ros-indigo-robot-localization-2.2.1-0trusty-20151111-0737/src/ros_filter.cpp:RosFilter<T>::transformPoseFailureCallback:1547
```
I have checked the tf tree. It seems OK.
[image]

Config for robot_localization is as this;
```
ekf_localization:
  frequency: 50
  odom0: /mobile_base_controller/odom
  odom0_config: [false, false, false,
                 false, false, false,
                 true, true, false,
                 false, false, false,
                 false, false, false]
  odom0_differential: false

  imu0: /imu/data_raw
  imu0_config: [false, false, false,
                false, false, true,
                false, false, false,
                false, false, true,
                false, false, false]

  imu0_differential: false
  odom_frame: odom
  base_link_frame: base_footprint
  world_frame: odom
  two_d_mode: true
```

Is this something related with TF frequencies? Should I decrease freq of ekf_localization? I could not figure it out. @ayrton04

ayrton04, 2016-02-02:
Are you running everything on the same machine? Try setting `use_tf_static` to true for `robot_state_publisher`.

akifh, 2016-02-02:
@ayrton04, thanks for the answer. I'm running all nodes on single machine on robot. Only `rqt_console` is running on a client machine.

`robot_state_publisher` publishes tfs based on URDF. After setting `use_tf_static` to `true`, warning seems disappeared. Now tf tree is like below;
 
[image]

Is it OK to have such a TF tree for URDF based configuration? What is the rationale for setting `use_tf_static` to `true`? If I remember right, with another IMU, I did not encounter with such a warning without `use_tf_static` option is set to `true`.

ayrton04, 2016-02-02:
Static transforms are treated as latched topics in tf2 so that you don't waste bandwidth publishing identical data at a high frequency. Since they don't change, having them be time-dependent doesn't really make sense.

The information you provided about a different IMU behaving differently suggests to me that the issue may be in your IMU driver. If I were you, I'd check the driver's source and make sure that the time stamp is being set immediately before the message is sent, rather than setting the time stamp, doing some work, and then publishing.

akifh, 2016-02-02:
To verify that IMU driver may have a problem, I switched to another IMU (from razor IMU to um6), it seems like with `use_tf_static` as `false`, I keep getting the tf warning. Only option seems like setting it to true. It makes sense to switch to static transform, though any comments on the cause of the issue? Thanks.

ayrton04, 2016-02-02:
The cause of the issue is that tf is failing to find a transform from your IMU frame to your base_link frame at the timestamp in the message. This means either (a) the time stamps are off for some reason, (b) the frame_id in the message itself is wrong, or (c) the transforms themselves are too old or too new. Since you have verified (a), and I can see for myself that (b) is correct, I have to assume (c). Looking at the robot_state_publisher wiki, it seems as though it future dates all fixed (static) transforms by 0.5 seconds:

http://wiki.ros.org/robot_state_publisher

Also, see [this](https://github.com/cra-ros-pkg/robot_localization/issues/248) related issue (read the whole thing).

akifh, 2016-02-03:
I also assume (c) is the reason after all these readings. This seems irrelevant with r_l, but I will further investigate the issue and provide feedback on the results. @ayrton04 , thank you for your help. Best.

closed 2016-02-03 by akifh