cra-ros-pkg/robot_localization #254: Interfacing with RTABMAP

zastrix, 2016-02-02:
Hello,

I am working to use robot_localization to take in info from a piksi gps, rp 2d lidar, and UM7 IMU and to give me a good estimate of where I am in space. I want to report this to rtabmap so I can relate this position to the 3d environment of my robot. When I run rtabmap without robot_localization everying works, however when I run robot_localization alongside rtabmap the odom frame drifts like crazy. I didn't know if this was an issue that you have seen before or not. Let me know what you need from me!

ayrton04, 2016-02-02:
Was this asked on ROS answers already?

zastrix, 2016-02-02:
I did not post it nor did I find someone that had :100:

ayrton04, 2016-02-02:
Would you mind posting it there? This board should really be for actual bugs or feature requests.

closed 2016-02-15 by ayrton04