cra-ros-pkg/robot_localization #262: Allow for the filter to operate on a fixed time-delay

NikolausDemmel, 2016-03-03:
AFAICT r_l currently has no explicit way to deal with delayed measurements (at least not if the delay is larger than the period of the filter update rate). Older measurements are simply applied to the current filter state as if their timestamp would match. In some situations this can be problematic (see e.g. http://answers.ros.org/question/227980/robot_localization-behavior-when-fusing-delayed-odometry-messages/).

One way to deal with this would be operating the filter at a fixed delay set in the configuration. That would mean, that measurements in the queue that are newer than `now() - filter_delay` are kept in the queue to be processed at a later time. This gives more time for the delayed measurements to arrive and be placed in the correct position in the queue.

Of course the downside would be that the pose estimate is available only at a delay, but depending on the application that might be ok, and you can tune the tradeoff between filter delay and accuracy. You could even run multiple instances of the filter, one delayed, but more accurate, and one undelayed, but less accurate.

There are other ways to handle measurement delays, but this is one that should be quick to implement, and incur little computational overhead (except for maybe maintaining a larger priority queue of measurements).

(P.S.: I hope it is ok to post these enhancement proposals / feature requests here. I don't expect for them to be worked on any time soon or ever, but I would like to discuss the principle, and hopefully I can also contribute some myself, as time allows).

ayrton04, 2016-03-03:
This is the ideal venue for these ideas, absolutely! Thanks for taking the time to add such detailed comments. This is something I heard from several users at ROSCon as well. Another possibility would be to store a buffer of state values and measurements going back _N_ seconds. When we receive an "old" measurement at time _t_, we start from the last cached state value before that time stamp, and then apply *all* measurements up to _t_. The only problem is that this may cause a sudden jump in the state estimate.

NikolausDemmel, 2016-03-03:
> Another possibility would be to store a buffer of state values and measurements going back N seconds.

I was just writing this up as #263. Good to see we are on the same page.

> The only problem is that this may cause a sudden jump in the state estimate. 

More than if the same measurement would arrive undelayed? I guess you are right, since the output of the filter (== the current filter state) is based more on pure prediction without (or with fewer) corrections. So both ideas could actually be combined to allow tuning the tradeoff between accuracy, time-delay and smoothness.

I guess when operating on a delay, there is much more that one could do (actually using all the  measurements in the queue, even the ones that have timestamp later than the current filter time). I would assume that this what is known as fixed-lag-smoothing with a whole body of its own literature (I know very little about it).

I suggested this simple variant mainly because it should be simple to implement.

essinger, 2016-03-06:
Hi there,

I stated the question on rosanswers.org and implemented a first draft for a solution to this problem based on the jade-dev branch on : https://github.com/essinger/robot_localization/tree/feat-temp-smoothing

The node now keeps a history of previous filter states and incorporated measurements. 

When encountering a measurement earlier than the current filter measurement time, it restores the oldest filter state before the delayed measurement and integrates ALL measurements up to the most recent including the delayed measurement.

You can specify the maximum supported delay in milliseconds by setting the history_filter_interval parameter. If set to 0 the feature should be completely off.

I tested it with my current problem and it seemed to work. Albeit with the cost that integrating so many measurements can cause the filter to drop in frequency significantly. 

I would appreciate some feedback to my suggested solution and wether there are some obvious faults with the code. Let me know!

ayrton04, 2016-03-07:
Thanks for taking the time to implement this. I'll take a look. One thing I will note is that the parameter should be given in seconds per [ROS standards](http://www.ros.org/reps/rep-0103.html#Units).

ayrton04, 2016-03-07:
Also @essinger, would you mind submitting a PR? It'll make reviewing and commenting easier. Looks pretty solid so far. @NikolausDemmel, if we have the more complex version described in #263, would you still be interested in having the more simplified version described here as well?

essinger, 2016-03-07:
Hey @ayrton04 ! My pleasure, I need this to work and your nodes already saved me so much time. I will open a Pull Request referencing this and issue 263.

ayrton04, 2017-02-21:
We have the ability to process delayed measurements as in #263. Closing. Please re-open if we want this method as well.

closed 2017-02-21 by ayrton04

NikolausDemmel, 2017-02-24:
Sounds good. Thanks!