cra-ros-pkg/robot_localization #264: Clarify documentation of how ..._config is applied

NikolausDemmel, 2016-03-03:
From the documentation on the wiki it is not 100% clear to me at what point the elements of measurements for which the `..._config` parameter is set to `false` are removed. Is this always with the directly with the incoming measurement before any processing, or does it happen after post-processing (i.e. frame transformations).

Concretley, for poses, is the selection applied in `header.frame_id`, or after transforming to `world_frame`. For twists, is the selection applied in `child_frame_id`, or after transforming to `base_link_frame`. What about the IMU message?

Whatever the answer, the explanation on the wiki should be clarified.

ayrton04, 2016-03-03:
Agreed. For the record, the data gets excluded _after_ it is transformed into the target `frame_id` (`world_frame` for pose data and `base_link_frame` for velocity data).

ayrton04, 2016-03-03:
This relates to #252.

NikolausDemmel, 2016-03-03:
> For the record, the data gets excluded after it is transformed into the target `frame_id`

I'm still not quite sure I understand.

Let use an example: consider a pose input where we concern ourselves for now only with x and y position. `header.frame_id` is `foo` and world frame is `odom`. The measured position is (1,0) in frame `foo` and rotated by 45° into frame `odom` such that its value is (1/√2, 1/√2). `pose_config` is "true, false, ...". What is the measurement with which the filter is updated?

ayrton04, 2016-03-04:
The application of the `_config` parameter always occurs once the data is in the target frame for the data type. For pose data, we first transform the incoming data into the `world_frame` frame, and then apply the `_config` setting (i.e., only the requested variables are passed into the actual filter). For twist data, we transform the incoming data into the `base_link_frame` frame, and then apply the `_config` setting.

So, in your example, the value that is given to the filter is just 1/√2 for X. The Y value is ignored.

NikolausDemmel, 2016-03-04:
What you say makes sense to me as the way it should be. But somehow the following code looks to me like the updateVector is actually rotated into the target frame somehow. In other words it looks to me as if the `_config` is applied to the input sensor data *before* transferring to target frame (I only looked at pose for now). Am I reading that wrong?

https://github.com/cra-ros-pkg/robot_localization/blob/bae0d53cd7824be78ea3169d0f500c9f1d53b9ac/src/ros_filter.cpp#L2309-L2349

ayrton04, 2016-03-04:
OK, so there are two aspects of handling the configuration: the transformation of the pose data, and the transformation of the update vector.

When I say that the config is applied after the transform is applied, what I mean is that I don't mask or zero out any values before they get transformed into their target frame. For example, assuming the same configuration you specified, if we received a measurement of (1, 1)  in the `foo` frame and had to rotate it 45 degrees, then we (1) apply the rotation and get (0, √2), and (2) apply the transformed config (more on that in a second). If I were to apply the config setting right away, then the measurement in the `foo` frame would become (1, 0), and then we'd apply the 45-degree rotation to get (1/√2, 1/√2), which would be wrong. My point is that we do the former: we first transform the source data into the target frame before the config setting is applied.

The second aspect of the config setting is that we rotate the update vector itself. In that sense, the configuration vector does indeed refer to the source data. This is because the user should not have to be concerned about how their transforms affect the measurement: the user knows that, for example, he or she only wants the X value in the measurement to make it into the filter. In that case, if we rotate by 45 degrees, the original X value will be split among X and Y in the transformed measurement*. Consider a sensor that only provides X velocity in the sensor frame, but comes from a sensor that is rotated by some number of degrees. In that case, the final measurement will have X and Y components, and it so we need to rotate the update vector so that we retain what the user requested, which is the source frame's X velocity value.

I apologize for not being clear before! I'm interested to hear your thoughts.

* It occurs to me that in the event that X and Y both have non-zero values, but we only want to keep X's value, we have to determine what component of the Y value makes it into the transformed final measurement and remove that. I think. I'd have to do the math on paper, but my intuition tells me that I'd need to handle that situation.

ayrton04, 2016-03-04:
Hmmm, the more I think about this, the less certain I have it right. I wonder if I do in fact need to zero out the components of the source-frame measurement that aren't being used, and then carry on with my update vector rotation logic as-is (for the same reason that a measurement in one axis may be projected on to other axes after rotation).

corot, 2016-09-28:
Hi, I contribute with my experience. I use the released packages for indigo.

In my scenario, I want to integrate the pose provided by amcl (green arrow) with another localization (white arrow) relative to the white line above the robot (obtained with distance sensors along the robot). 
This pose is relative to the tf above the white line and only the distance to the line (coordinate y) and the relative angle matters. So for x I just provide the value obtained from amcl. And on poseXXX_config I have:
```
pose2_config: [false, true,  false,
               false, false, true,
               false, false, false,
               false, false, false,
               false, false, false]
```

[image]

This made sense to me because, according to the  wiki, the poseXXX_config is applied **before** transforming to world frame! And whatever value comes in the x coordinate should be ignored. But it is not. If I add an artificial +20cm, the robot runs forward:

[animated gif](https://drive.google.com/file/d/0BzYjlgbSZJSWSlQ5c3BsVkxVR1E/view?usp=sharing)

So either I'm doing something wrong, or there's a problem with poseXXX_config. Hope I explained my case properly. If I set x covariance to an astronomic value, I obtain the desired result, but I suppose the role of poseXXX_config is to make this much cleaner.

Btw, thanks for contributing this amazing software!

ayrton04, 2016-09-28:
Thanks! I'll look into this.

ayrton04, 2016-10-11:
@corot, can you please post a bit more information? Perhaps you launch/config files, sample input messages, and any transforms between frames?

ayrton04, 2017-02-28:
So I can't really decide whether it's better to apply the sensor configuration in the source or target frames, so I will just add an option for both and let users decide.