cra-ros-pkg/robot_localization #267: Invalid integration of Z pose

plnegre, 2016-03-11:
Hello!

I'm using the ekf_node with the following configuration:

`
<param name="twist0" value="/sensors/dvl"/>
    <param name="imu0" value="/sensors/imu"/>
    <param name="pose0" value="/sensors/depth_new"/>

    
    <rosparam param="twist0_config">[false, false, false,
                                     false, false, false,
                                     true,  true,  true,
                                     false,  false,  false,
                                     false, false, false]</rosparam>
    <rosparam param="imu0_config">[false, false, false,
                                   true, true, true,
                                   false, false, false,
                                   true, true, true,
                                   false, false, false]</rosparam>
    <rosparam param="pose0_config">[false, false, true,
                                    false, false, false,
                                    false, false, false,
                                    false, false, false,
                                    false, false, false]</rosparam>
`

This is for an AUV. I'm using 3 sensors: imu, a DVL which gives me linear velocities and a pressure sensor which gives me position in Z.

The problem is with the pose message (pressure), which seams to be not well integrated. If I open the diagnostics I see the following error:

`
Full Name: /EKF Localization/ekf_localization: Filter diagnostic updater
Component: ekf_localization: Filter diagnostic updater
Hardware ID: none
Level: WARNING
Message: Potentially erroneous data or settings detected for a robot_localization state estimation node.

pose0_covariance: The covariance at position (7), which corresponds to Y variance, was zero. This will be replaced with a small value to maintain filter stability, but should be corrected at the message origin node.
`

Of course, the covariances at position 0 and 7 are 0.0, because I don't have x and y measures. But I clearly specified this on the param "pose0_config".

Any suggestions?

ayrton04, 2016-03-12:
What's the `frame_id` of the Z data? Can you pose the entire launch file and a sample message of each?  Is the filter not accepting the measurement, or are you just seeing the diagnostic and it's otherwise working correctly? I've run the EKF with this exact setup (DVL, IMU, pressure sensor) and it worked well.

plnegre, 2016-03-13:
Hello @ayrton04 

this is my launch file:

`
<node pkg="robot_localization" type="ekf_localization_node" name="ekf_localization" output="screen" clear_params="true">

    <param name="frequency" value="30"/>
    <param name="sensor_timeout" value="0.1"/>

    <param name="map_frame" value="map"/>
    <param name="odom_frame" value="odom"/>
    <param name="base_link_frame" value="robot"/>
    <param name="world_frame" value="odom"/>

    <param name="twist0" value="/sensors/dvl"/>
    <param name="imu0" value="/sensors/imu"/>
    <param name="pose0" value="/sensors/depth"/>

    <param name="imu0_remove_gravitational_acceleration" value="true"/>

    
    <rosparam param="twist0_config">[false, false, false,
                                     false, false, false,
                                     true,  true,  true,
                                     false,  false,  false,
                                     false, false, false]</rosparam>
    <rosparam param="imu0_config">[false, false, false,
                                   true, true, true,
                                   false, false, false,
                                   true, true, true,
                                   false, false, false]</rosparam>
    <rosparam param="pose0_config">[false, false, true,
                                    false, false, false,
                                    false, false, false,
                                    false, false, false,
                                    false, false, false]</rosparam>
</node>
`

All other parameters are set to default. I also tried setting <param name="pose0_differential" value="true"/> but it does not works.

Simply, the output of the ekf is wrong when I introduce the depth message (it gives me a position totally incoherent). I also tried with imu and dvl and it is working fine.

The sample messages are:

IMU:
`
header: 
  seq: 2220
  stamp: 
    secs: 1457696098
    nsecs: 297723054
  frame_id: robot
orientation: 
  x: -0.00979521632193
  y: 0.059209976992
  z: 0.490810674875
  w: 0.869196821088
orientation_covariance: [0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]
angular_velocity: 
  x: -0.0750247618653
  y: -0.0303648026509
  z: -0.051172665286
angular_velocity_covariance: [0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]
linear_acceleration: 
  x: 1.03862158822
  y: -0.404053038694
  z: -9.80592843527
linear_acceleration_covariance: [0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]
`

DVL:
`
header: 
  seq: 1297
  stamp: 
    secs: 1457696139
    nsecs: 419182656
  frame_id: dvl
twist: 
  twist: 
    linear: 
      x: -0.287
      y: 0.27
      z: -0.005
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.002, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.002, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.002, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
`

DEPTH:
`
header: 
  seq: 3250
  stamp: 
    secs: 1457696170
    nsecs: 494995519
  frame_id: pressure
pose: 
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.172176214609
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.3, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
`

Of course, I'm publishing the static TFs between all frames.

Thank you very much!

ayrton04, 2016-03-13:
Ah, I think I see your problem. Your depth message is reported in the "pressure" frame. Am I correct in assuming that you are broadcasting a *base_link*->*pressure* transform? If so, that's your issue. All pose data is transformed into the `world_frame` before we fuse it. If your frame for your pressure data is relative to *base_link* (*robot* for you), the EKF will transform it from *sensor* to *robot*, **then** to *odom*, which is the very transform that the EKF is supplying. You are effectively adding the current vehicle state to every Z measurement that arrives, which means that after measurement *N*, you are getting something like *Z* * *N* for your depth measurement.

The solution is to make the *pressure* frame a child frame of *odom*, not *robot*. However, you'll have to make it a dynamic transform (i.e., you'll need another node) as the distance between the depth sensor and the vehicle origin will change with pitch and roll. Of course, if your depth sensor is at the vehicle origin, then you're set.

plnegre, 2016-03-14:
Thank you @ayrton04  for your fast reply! We are close, but there are some points I don't understand...

First of all, you say: "You are effectively adding the current vehicle state to every Z measurement that arrives, which means that after measurement N, you are getting something like Z * N for your depth measurement." Yes, this is what happens.

My pressure sensor is not on the robot origin. Should I provide a dynamic TF between odom and pressure? And a fixed TF between pressure and robot? Which TFs should I broadcast and which frame_id should I set to the pressure message?

Thank you very much.

ayrton04, 2016-03-14:
**No** transform should exist between *robot* and *pressure*. You need to generate an *odom*->*pressure* transform, and that transform will have to be dynamic. The messages themselves can remain as-is.

plnegre, 2016-03-15:
Ok, solved! Working cool now!

Thank you very much.

closed 2016-03-15 by plnegre

Tdoe4321, 2021-03-18:
I don't mean to dig up this 5 year old thread, but I'm running into the exact same issue. I'm running all the same sensors on my AUV and the same launch configs, but I'm setting all the sensor frame_id's to 'base_link'. It is my understanding that I don't need to generate transforms if this is the case, correct?

My Z filtered value is blowing up, even with a static z measurement value.

ayrton04, 2021-03-22:
This should probably have been a ROS Answers question, but as it's already here...

The sensor needs to be reporting data in the coordinate frame that you specify. If you tell the filter that a given sensor is in the `base_link` frame, then the data should be in that coordinate frame. A depth sensor is clearly not in the `base_link` (body) frame of the robot. It is relative to the surface of the sea.

Let's say your depth sensor reports a depth of `-5.0` in a given cycle. If that is reported as being in the `base_link` frame, you are telling the filter that it just got a measurement 5 meters below _where it is now_. So in the next cycle, you give it another measurement of `-5.0`, so the filter thinks you moved 5 meters further down again. The state estimate will just rocket straight downward.

If your AUV were always level, or the sensor was positioned exactly at the center of mass of your AUV, then this would be pretty straightforward: just report the depth sensor measurement in, say, the `map` or `odom` frame (whatever your world frame is). So then the filter output would just produce -5.0 for the vehicle's Z coordinate.

The problem is when the sensor is offset from the vehicle centroid. Now you have a bigger problem, and one that `r_l` doesn't support. You need to determine where the sensor is in the base_link frame, but then _transform that_ into the `odom` frame, and then add that to the depth measurement. So if your vehicle were pitched very strongly downward, the vehicle might report, say, -5.4 meters. But the vehicle centroid is still just at -5.0.