cra-ros-pkg/robot_localization #271: Example launch file for using multiple robot_localization ekf nodes

baronep, 2016-03-18:
I have seen a lot of people writing about using multiple ekf nodes (and a navsat node) for odom & map localization. Would it be possible to put together an example launch file for configuring both of these nodes together?

ayrton04, 2016-03-24:
Yes, I'll try to put something together soon.

bitmorse, 2016-07-03:
+1 on this

ayrton04, 2016-08-05:
This has been pushed to the 281-documentation branch, if you'd like to take a look.

bitmorse, 2016-08-05:
thanks a lot! I've been having trouble getting a stable output from navsat_transform_node or ekf_gps (i'm not sure which one is exploding due to the cyclical nature). I used your previous example initial covariances (https://github.com/Octanis1/Octanis1-ROS/blob/master/catkin_ws/src/octanis1_description/params/robot_localization_ekf_gps.yaml) which are different to the one's you posted now. Is there any reason for the difference?

ayrton04, 2016-09-05:
Sorry, that link is broken. I tend to inflate the initial covariance for any variable that I'm measuring, which ensures that the filter will converge more rapidly to the first measurements from the sensor. If the estimate variance for any given variable is too small, then the filter trusts its own value more than the measurement, and will be slow to "accept" it.

closed 2016-10-24 by ayrton04