cra-ros-pkg/robot_localization #276: Subscribe data with tcpNoDelay()

jbler, 2016-04-27:
Hi,

I am running the EKF at 200 Hz at that frequency I am not able to get every spin a new measurement but my subscribing measurement is sent a 200 Hz. Instead my data is often arriving at bunches of a few measurments together and therefore with a lower frequency.  

Would it be possible to subscribe a topic  using the ros::TransportHints().tcpNoDelay() option which seems a reasonable try for me.

ayrton04, 2016-04-27:
Have you tried increasing the queue size for the sensor in question?

ayrton04, 2016-04-27:
But the short answer is that yes, I think it would be a good idea. Would you be willing to fork the repo and try it?

jbler, 2016-04-27:
Thanks for the fast reply. 
Sure, I will give it a try later!

closed 2016-05-09 by ayrton04