cra-ros-pkg/robot_localization #281: Lost Documentation in Example Launch Files

Jmeyer1292, 2016-05-11:
Recently a [commit](https://github.com/cra-ros-pkg/robot_localization/commit/4392935a6530e9f2da0e8cff4d54dcf9f41b8fe7) simplified the launch file by moving parameters to entirely separate file. In the process, a lot of incredibly useful documentation on the meaning of these parameters for novices like myself was lost. I realize that much of this information is available on your wiki page, but I thought the inline documentation was really great and a sign of tremendous quality. If I get some time, I'll try to transpose the old stuff to the new yaml file.

Additionally, is there any documentation on the new control component that was recently merged?

Thanks for a wonderful package.

ayrton04, 2016-05-11:
Sure, we can put those back. You're not the first one to request it! :) I'll also update both those and the wiki to include the new control term and lagged measurement functionality.

If you want to submit a PR for adding the documentation back to the template config files, that would be great. No worries if you don't have time!

asimay, 2016-06-02:
hi dear @ayrton04 , with regard to param file:

control_config: [true, false, false, false, false, true]
acceleration_limits: [1.3, 0.0, 0.0, 0.0, 0.0, 3.4]

does the order stand for vx, vy, vz, ax, ay, az?
thank you for your clarification!

ayrton04, 2016-08-05:
This has been pushed to the 281-documentation branch, if you'd like to take a look.

closed 2016-10-24 by ayrton04