cra-ros-pkg/robot_localization #282: navsat_transform_node not longer able to calculate local GPS position

tik0, 2016-05-26:
ROS: Indigo
Affected version: 2.2.3

Since the upgrade from 2.2.2 to 2.2.3, navsat_transform_node just throws this warning, while the calculated pose remains always at (0,0,0):
`[ WARN] [1464301493.207970288, 1435140992.189921469]: Could not obtain transform from world to GPS. Error was "world" passed to lookupTransform argument source_frame does not exist.`
I think it is clear that their cannot exist some frame like "world", because this is why we are using this node, to get some child frame named "utm" which refers to the real world's origin.

So why do you now care about the parent frame in the message, and how should I setup my tf-tree to make it work again?

ayrton04, 2016-05-26:
The "world" frame would get set to the `frame_id` of the odometry message that you are using as input. Is the warning message you posted the exact text you're seeing? Also, can you post sample input messages for everything, along with your `navsat_transform_node` config? That makes it a bit easier to determine what's going on. Thanks!

ayrton04, 2016-05-26:
There are only four lines in `navsat_transform_node` that can generate that error:

https://github.com/cra-ros-pkg/robot_localization/blob/2.2.3/src/navsat_transform.cpp#L349
https://github.com/cra-ros-pkg/robot_localization/blob/2.2.3/src/navsat_transform.cpp#L392
https://github.com/cra-ros-pkg/robot_localization/blob/2.2.3/src/navsat_transform.cpp#L401
https://github.com/cra-ros-pkg/robot_localization/blob/2.2.3/src/navsat_transform.cpp#L477

In all of them, the target frame (second parameter) is `base_link_frame_id_`. Given the error message you've posted, it looks to me like you have your `base_link` frame is `GPS`. Is that correct?

tik0, 2016-05-27:
I have a vehicle base link called "base_link" with an static attached frame called "GPS".
"robot_localization" provides the transform from "odom" to "base_link".
Inside the NavSatFix message, "world" is the parent frame, and "GPS" the child.
Before, "world" was just a placeholder so I never used it, and "navsat_transform_node" publishes the transform "odom->GPS" as expected.

Now, "navsat_transform_node" is complaining about "world", but isn't it then braking the conventions of re-parenting?
I mean, it assumes that there is the TF "world->GPS" and it would publish "odom->GPS", right?

I'll provide more information as soon as possible.

ayrton04, 2016-06-01:
So the NavSatFix message does not have a child frame. It just has a header field, with the normal `frame_id`:

http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html

What I suspect is happening is that the old version of `navsat_transform_node` didn't take into account the sensor's offset from the vehicle origin, so the pose you were getting out of it was technically incorrect. Now I take the `base_link->gps` offset, rotate it by the robot's current orientation (i.e., from the odom->base_link transform), and then correct the final pose with that value. I'm not sure why you're having this trouble, but again, I'd have to see sample input messages and your full launch file(s).

ayrton04, 2016-08-01:
Can you try this with the latest code base and let me know if it's resolved? Just make sure the transform from the GPS frame to the base_link frame exists. If it's not resolved, can you post your inputs here? Thanks!

ayrton04, 2017-02-21:
I'm assuming this has been fixed? I will close. Please re-open if this is still a concern.

closed 2017-02-21 by ayrton04