cra-ros-pkg/robot_localization #283: after update to new code, roslaunch occurs python running error..

asimay, 2016-06-01:
after update to new code, roslaunch occurs python running error..
old version has no such warning.
if i comment the include" ekf launch file", the warning disappeared. so some params reading maybe have some issues.
[image]

asimay, 2016-06-01:
launch file:
[image]

ayrton04, 2016-06-01:
Are you using the template directly, or have you modified it? Any chance you can try removing parameters until the warning disappears?

asimay, 2016-06-02:
hi, dear @ayrton04 , I re-download the new version and use new version's code totally, the warning disappered. but actually my version looks no difference with new version except a little of comments, and i used binary tool "bccompare" to look up the difference, but find nothing. it is very strange. but anyway,  Thank you very much!

closed 2016-06-02 by asimay