cra-ros-pkg/robot_localization #284: in  new version, velocity & acceleration's filter sometimes caused robot collide to the wall.

asimay, 2016-06-02:
in  new version, velocity & acceleration's filter rules sometimes caused robot collide to the wall.
if inhibit robot_localization module, the robot could definitely turned around and arrived at aim safely.
BTW: I used navigation stack to simulate.

Please see below link, especially sim time at 38 sec and at last:
https://youtu.be/pbxcLVV57rs

Please help to check. thank you very much!

ayrton04, 2016-06-02:
Looking at the video, it appears that your local planner is planning a path through the wall. Your robot is not colliding with the wall. Furthermore, this behavior would likely have nothing to do with `robot_localization`. The EKF node in `robot_localization` is simply estimating where your robot is. In your `rviz` display, this is the single, large red arrow that is attached to your robot. Note that it is not moving when it shouldn't be, nor is its pose estimate inaccurate when compared to the ground truth on the left.

In the future, it might be better to post questions on http://answers.ros.org and confirm that there's a bug before posting an issue here.

closed 2016-06-02 by ayrton04

asimay, 2016-06-07:
hi, dear @ayrton04 ,i find some issues when combined with robot_localization EKF, this maybe well illustrated the reason sometimes my car acts not very stable when combined with EKF & stageros.
when i run roswtf:
[image]

EKF and stage publish same tf transform, so sometimes the behavior is not very stable, but can work.
if under stage env, we maybe need to exclude robot_location package (because there is no switch of tf publish in robot_localization package).
it gets good result as I wish.
Let you know. Thanks!

ayrton04, 2016-06-07:
Two things:

1. You can _not_ have two nodes publishing the same transform. You should configure `r_l` or `stage` to not broadcast the _odom_->_base_link_ transform.
2. You can turn off the `tf` broadcast in `r_l` using the [publish_tf parameter](https://github.com/cra-ros-pkg/robot_localization/blob/indigo-devel/src/ros_filter.cpp#L650).

asimay, 2016-06-07:
hi, dear @ayrton04 , Thank you very much! Thank you for your patient instruction! I'm really appreciate your help!

ayrton04, 2016-06-07:
No problem! I would recommend posting on ROS Answers whenever possible, as there are a large number of people - including members of OSRF - who can answer pretty much any question you may have.

asimay, 2016-06-12:
hi, dear @ayrton04 , I asked simular question about r_l & amcl & navsat migration on ROS Answers:
http://answers.ros.org/question/236403/tf-problems-when-combine-robot_localization-gps-navigation-amcl-stack/
but get no feedback till now, could you please instruct me on this issue? I'm very appreciate your help! Thank you very much!