cra-ros-pkg/robot_localization #288: Problem when fusing IMU and Odometry

tinokl, 2016-06-06:
Hi, I am trying to use robot localization to fuse the data from an XSense IMU and Odometry Source (Pioneer). The output is not what i expect actually. The resulting odometry looks similar (almost exact) to the odometry of the pioneer? So the filter is not improving or at least changing the result?

This is my launch file:

    <param name="frequency" value="30"/>
    <param name="sensor_timeout" value="0.1"/>

    <param name="base_link_frame" value="base_footprint"/>
    <param name="two_d_mode" value="true"/>

    <param name="imu0" value="imu/data"/>
    <rosparam param="imu0_config">[false, false, false,
                                   false, false,  true,
                                   false, false, false,
                                   false, false, false,
                                   false,  false, false]</rosparam>
    <param name="imu0_differential" value="true"/>
    <param name="imu0_remove_gravitational_acceleration" value="false"/>

    <param name="odom0" value="/odom/fixed_wheel"/>
    <rosparam param="odom0_config">[true, true,  false, 
                                    false, false, true, 
                                    false, false, false, 
                                    false, false, false,
                                    false, false, false]</rosparam>
   <param name="odom0_differential" value="false"/>

      <param name="odom0_relative" value="true"/>
      <param name="imu0_relative" value="true"/>

Maybe i am missing something obvious?
Thanks in advance!

ayrton04, 2016-06-06:
Please ask this question on [answers.ros.org](answers.ros.org). If you've already asked it, please point me to it. Thanks!

ayrton04, 2016-06-06:
P.S. - Also, please post sample input messages, and not just your launch file config.

closed 2016-06-06 by ayrton04

tinokl, 2016-06-07:
Thanks for the fast reply, a added the question on ROS answers and added bag files
[Ros Answers](http://answers.ros.org/question/236161/problem-when-fusing-imu-and-odometry/)