cra-ros-pkg/robot_localization #290: Navsat_transform_node never take imu_topic into account.

asimay, 2016-06-14:
snapshot as belows:
Navsat_transform_node never take imu_topic into account. 
[image]

```
      <remap from="/imu/data" to="imu_topic"/>
      <remap from="/gps/fix" to="imu_nav_topic"/>
      <remap from="/odometry/filtered" to="odom_filtered_gps"/>
```
and no matter I set:
`<param name="use_odometry_yaw" value="true"/> `
or
`<param name="use_odometry_yaw" value="false"/>`

imu_topic is not taken into account.

ayrton04, 2016-06-14:
Again, this should be a ROS Answers question, and we should verify that there is a problem before posting an issue. In this case, the IMU subscription is only needed until a single IMU message is received, at which point the subscriber is taken down:

https://github.com/cra-ros-pkg/robot_localization/blob/indigo-devel/src/navsat_transform.cpp#L198

closed 2016-06-14 by ayrton04

asimay, 2016-06-14:
hi, dear @ayrton04 : 
Thank you very much! But when I set 
`<param name="use_odometry_yaw" value="false"/>`
terminal occurs such errors: 

[image]

ayrton04, 2016-06-14:
Looks like a ROS answers question. Please post your configuration and sample input messages.