cra-ros-pkg/robot_localization #298: Manual Datum Does not support setting gps frame, so it is ""

bcfjohns, 2016-06-29:
The manual datum topic allows setting of the base link and world frames, but not the gps frame, so when the  node is used with only a manual datum, the gps frame is "". This means the node constantly sends warnings and then publishes odometery information as all zeros.

ayrton04, 2016-06-30:
Thanks! Any chance you'd be willing to take a crack at a PR?

bcfjohns, 2016-07-08:
Unfortunately, not anytime soon.

skohlbr, 2016-07-12:
+1, just ran into this also. For setting datum manually, adding a frame_id as param seems possible. Or should we wait for first fix data and then use that? There are a few options of adding this, but they have some subtle differences. Setting via the service should also work correctly.

ayrton04, 2016-07-12:
Can you try this fork and let me know if it's fixed?

https://github.com/locusrobotics/robot_localization

I plan to merge upstream. Part of the problem is that I'd have to add another parameter to the datum to fix this issue, and after some consideration, it seems that I ought to still use the frame_ids from the input messages, rather than forcing users to add them to the datum. What do you think?

ayrton04, 2016-07-15:
Has anyone had a chance to try this?

ayrton04, 2016-08-01:
@bcfjohns @skohlbr I think I've addressed this, but I'd have to see how it works for you. I had to change the datum parameter behavior so that it doesn't concern itself with frame_ids; those are now assumed to come from the input messages, even if the datum is specified.

ayrton04, 2016-09-07:
Closing this. Feel free to reopen if it's still a problem.

closed 2016-09-07 by ayrton04