cra-ros-pkg/robot_localization #301: Zigzag ground track from eight shape flight path

tdnet12434, 2016-08-22:
Hi all, 
I tried to using ukf template recursively with navsat_transform node following http://answers.ros.org/question/200071/how-to-fuse-imu-gps-using-robot_localization/

With my eight flight path it seem zigzag ground track like due to false orientation of Accelerometer.
This is my setup launch file.

```
<launch>

    <node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true" output="screen">

      <!-- Frequency of the main run loop. -->
      <param name="frequency" value="30"/>

      <!-- Delay time, in seconds, before we calculate the transform from the UTM frame to your
           world frame. This is especially important if you have use_odometry_yaw set to true.
           Defaults to 0. -->
      <param name="delay" value="0"/>

      <!-- PLEASE READ: Like all nodes in robot_localization, this node assumes that your
           IMU data is reported in the ENU frame. Many IMUs report data in the NED frame,
           so you'll want to verify that your data is in the correct frame before using it.  -->

      <!-- If your IMU does not account for magnetic declination, enter the value for your
           location here. If you don't know it, see http://www.ngdc.noaa.gov/geomag-web/
           (make sure to convert the value to radians). This parameter is mandatory. -->
      <param name="magnetic_declination_radians" value="0"/>

      <!-- Your IMU's yaw, once the magentic_declination_radians value is added to it, should
           report 0 when facing east. If it doesn't, enter the offset here. Defaults to 0. -->
      <param name="yaw_offset" value="0"/>

      <!-- If this is true, the altitude is set to 0 in the output odometry message. Defaults
           to false. -->
      <param name="zero_altitude" value="true"/>

      <!-- If this is true, the transform world_frame->utm transform is broadcast
           for use by other nodes. Defaults to false. -->
      <param name="broadcast_utm_transform" value="false"/>

      <!-- If this is true, all received odometry data is converted back to a lat/lon
           and published as a NavSatFix message as /gps/filtered. Defaults to false. -->
      <param name="publish_filtered_gps" value="false"/>

      <!-- If this is true, the node ignores the IMU data and gets its heading from the
           odometry source (typically the /odometry/filtered topic coming from one of
           robot_localization's state estimation nodes). BE CAREFUL when using this!
           The yaw value in your odometry source *must* be world-referenced, e.g., you
           cannot use your odometry source for yaw if you have only one source for yaw
           data and it has its differential or relative parameter set to true. Defaults
           to false. -->
      <param name="use_odometry_yaw" value="false"/>

      <!-- If true, will retrieve the datum from the 'datum' parameter below, if available.
           If no 'datum' parameter exists, navsat_transform_node will wait until the user
           calls the 'datum' service with the SetDatum service message. -->
      <param name="wait_for_datum" value="false"/>

      <!-- Instead of using the first GPS location and IMU-based heading for the
      local-frame origin, users can specify the origin (datum) using this parameter.
      The fields in the parameter represent latitude and longitude in decimal degrees,
      and heading in radians. As navsat_transform_node assumes an ENU standard, a 0
      heading corresponds to east. -->
      <!-- <rosparam param="datum">[7.006354, 100.502297, 0.0, odom, fcu]</rosparam> -->

      <!-- Placeholders for input remapping. Set your topic names as the "to" values.
      <remap from="/imu/data" to=""/>
      <remap from="/odometry/filtered" to=""/>
      <remap from="/gps/fix" to=""/>
      -->
      <remap from="/imu/data" to="/mavros/imu/data"/>
      <remap from="/gps/fix" to="/mavros/global_position/raw/fix"/>
      <!-- Placeholders for output remapping.
      <remap from="/odometry/gps" to=""/>
      <remap from="/gps/filtered" to=""/>
      -->

    </node>

  <node pkg="robot_localization" type="ukf_localization_node" name="ukf_se" clear_params="true">
    <rosparam command="load" file="$(find robot_localization)/params/ukf_max.yaml" />

    <!--  Placeholder for output topic remapping
    <remap from="odometry/filtered" to=""/>
    -->

  </node>
</launch>
<launch>

    <node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true" output="screen">

      <!-- Frequency of the main run loop. -->
      <param name="frequency" value="30"/>

      <!-- Delay time, in seconds, before we calculate the transform from the UTM frame to your
           world frame. This is especially important if you have use_odometry_yaw set to true.
           Defaults to 0. -->
      <param name="delay" value="0"/>

      <!-- PLEASE READ: Like all nodes in robot_localization, this node assumes that your
           IMU data is reported in the ENU frame. Many IMUs report data in the NED frame,
           so you'll want to verify that your data is in the correct frame before using it.  -->

      <!-- If your IMU does not account for magnetic declination, enter the value for your
           location here. If you don't know it, see http://www.ngdc.noaa.gov/geomag-web/
           (make sure to convert the value to radians). This parameter is mandatory. -->
      <param name="magnetic_declination_radians" value="0"/>

      <!-- Your IMU's yaw, once the magentic_declination_radians value is added to it, should
           report 0 when facing east. If it doesn't, enter the offset here. Defaults to 0. -->
      <param name="yaw_offset" value="0"/>

      <!-- If this is true, the altitude is set to 0 in the output odometry message. Defaults
           to false. -->
      <param name="zero_altitude" value="true"/>

      <!-- If this is true, the transform world_frame->utm transform is broadcast
           for use by other nodes. Defaults to false. -->
      <param name="broadcast_utm_transform" value="false"/>

      <!-- If this is true, all received odometry data is converted back to a lat/lon
           and published as a NavSatFix message as /gps/filtered. Defaults to false. -->
      <param name="publish_filtered_gps" value="false"/>

      <!-- If this is true, the node ignores the IMU data and gets its heading from the
           odometry source (typically the /odometry/filtered topic coming from one of
           robot_localization's state estimation nodes). BE CAREFUL when using this!
           The yaw value in your odometry source *must* be world-referenced, e.g., you
           cannot use your odometry source for yaw if you have only one source for yaw
           data and it has its differential or relative parameter set to true. Defaults
           to false. -->
      <param name="use_odometry_yaw" value="false"/>

      <!-- If true, will retrieve the datum from the 'datum' parameter below, if available.
           If no 'datum' parameter exists, navsat_transform_node will wait until the user
           calls the 'datum' service with the SetDatum service message. -->
      <param name="wait_for_datum" value="false"/>

      <!-- Instead of using the first GPS location and IMU-based heading for the
      local-frame origin, users can specify the origin (datum) using this parameter.
      The fields in the parameter represent latitude and longitude in decimal degrees,
      and heading in radians. As navsat_transform_node assumes an ENU standard, a 0
      heading corresponds to east. -->
      <!-- <rosparam param="datum">[7.006354, 100.502297, 0.0, odom, fcu]</rosparam> -->

      <!-- Placeholders for input remapping. Set your topic names as the "to" values.
      <remap from="/imu/data" to=""/>
      <remap from="/odometry/filtered" to=""/>
      <remap from="/gps/fix" to=""/>
      -->
      <remap from="/imu/data" to="/mavros/imu/data"/>
      <remap from="/gps/fix" to="/mavros/global_position/raw/fix"/>
      <!-- Placeholders for output remapping.
      <remap from="/odometry/gps" to=""/>
      <remap from="/gps/filtered" to=""/>
      -->

    </node>

  <node pkg="robot_localization" type="ukf_localization_node" name="ukf_se" clear_params="true">
    <rosparam command="load" file="$(find robot_localization)/params/ukf_max.yaml" />

    <!--  Placeholder for output topic remapping
    <remap from="odometry/filtered" to=""/>
    -->

  </node>
</launch>

```

and this is my dataset
https://1drv.ms/u/s!Amsv72jI4yJwgS-R1PYE6r0e72F3

it's start at 180s so this command will help.

`rosbag play px4_eight_football_2016-07-22-14-48-23.bag --clock -s 180`

Thank you advanced.

tdnet12434, 2016-08-23:
Hello @ayrton04 Can you help me please.

ayrton04, 2016-08-23:
Did you ask this on ROS answers? If so, can you please direct me to it? This board is really for bugs and issues. Answering these questions takes a lot of time, so I may not get to it immediately.

tdnet12434, 2016-08-23:
Sorry for my wrong understand about purpose of board and thank you for direction. Here is my topic on ROS forum. http://answers.ros.org/question/242243/zigzag-ground-track-from-eight-shape-flight-path/

So i close this issue now.

closed 2016-08-23 by tdnet12434