cra-ros-pkg/robot_localization #302: navsat_transform produces odometry that changes with heading

msy22, 2016-09-03:
I think I've identified a bug within navsat_transform. The odometry has a systematic error of up to 30 cm that is dependent on the magnetic heading of the robot. I think the error was introduced in one of the changes between version 2.2.2 and 2.2.3, because the error only appear after I updated my robot_localization version. I think the error is being cause by navsat_transform improperly handling a frame transform when transforming the UTM position to a pose relative to the /map origin. Possibly the /utm -> /map transform. 

The error is easiest to identify if you simply rotate the robot on the spot in an outdoor environment, as shown:

[image]

The "GPS" stream is from navsat_transform, and the "AUXGPS" stream is from a node I wrote myself which performs the same function. The "Local" stream is the output from an instance of ekf_localization fusing wheel odometry and IMU data. "Global" fuses IMU, odometry and gps data from a high-accuracy (~20mm) gps receiver.

The problem is documented in full [here](http://answers.ros.org/question/242015/why-does-the-accuracy-of-navsat_transform-change-with-heading/?comment=242957#post-id-242957).

ayrton04, 2016-09-03:
Answered your (very thorough!) ROS answers question. I'm suspecting this has to do with the base_link->gps transform, but I need more information to verify. The bag you provided will help, thanks!

ayrton04, 2016-09-05:
Two questions:

1. In your graph, you show the GPS plot moving up and down. Did you actually record and plot the `/navsat/auxgps/odometry` topic, or use the 

*map->base_link* transform?
2. Perhaps I'm missing what you did, but is your "Global" plot fusing the output of `n_t_n` or your node?

The reason I ask about (1) can be seen in [this](https://drive.google.com/open?id=0BzZeubT6hZReQWlfV3Q4STl5R28) video, in which I am playing back your latest in-place rotation bag file along with the launch file you used. Note that I am using version 2.3.0 downloaded from source. The robot's *map->base_link* transform does indeed move up and down along a single axis (although it's Y for me). However, the `/navsat/auxgps/odometry` is shown in yellow, and it stays within a very small area, and its motion doesn't seem to correlate strongly with the rotation of the robot.

ayrton04, 2016-09-05:
Can you please try the `302-fix-offset-handling` branch and let me know if it's been fixed? Thanks!

msy22, 2016-09-05:
Sorry about the confusion. Yes it looks like you were using the original launch file which fused `/odometry/gps` in the global instance of `ekf_localization` (the output of which is the "Global" data on the graph). In my current launch file I fuse `/navsat/auxgps/odometry` instead. 

I see the key difference with `302-fix-offset-handling` is that the order of `world_frame_id_` and `base_link_frame_id_` has been reversed. I'll give it a shot and report back.

ayrton04, 2016-09-06:
The issue was pretty straightforward. As I said on ROS Answers, the only reason I need the robot's world-frame pose at that point is that I need its orientation. If I have that, I can take the *base_link*->*gps* offset, rotate it so that it is also in the world frame, then remove it from the GPS-in-robot's-world-frame-pose, thus producing the location of the robot's centroid.

The problem was that I was rotating it the wrong way, so your offset of `-0.12` in robot-frame `X` was getting doubled and added to *world-frame* `Y` when the robot's heading was `pi/2`. The offset would remain at `0` for headings of `0` and `pi`. This explains both the magnitude and behavior of the error you were seeing.

ayrton04, 2016-09-06:
[Here's a link to a video](https://drive.google.com/open?id=0BzZeubT6hZReMHAzekZBVmpCMFE) in which I ran your launch and bag file, but did *not* fuse the output of `n_t_n` back into the EKF (so I commented out `odom1`). This allows us to see if the state estimate GPS data agree.

Incidentally, for some reason, your bag file freezes up around the 57-second mark, and then resumes. This is why you see the strange behavior at the second corner.

msy22, 2016-09-06:
Ah right, that would make sense. It also explains why the node I wrote doesn't have that problem - because it never actually uses the /base_link -> /auxgps_link transform.

And well done spotting that the maximum error was approximately twice the linear X offset. Given how much time I've spent staring at that one graph I really should have picked up on that. 

As for the bagfile freezing, I'm not sure what's causing that sorry. There's still a few odd things that sometimes happen with my setup. Most of it is related to the IMU bugging out or the hardware struggling to run everything fast enough (my map size is quite big for long-distance outdoor navigation).

ayrton04, 2016-09-07:
No worries! Just thought I'd point out why the video seemed to jump there.

I've merged the PR and will close this. Thanks very much for such a thorough investigation and for reporting this issue!

closed 2016-09-07 by ayrton04