cra-ros-pkg/robot_localization #305: Bug report. NAV module always prompt "Could not obtain transform from odom->baselink"

asimay, 2016-09-21:
error prompt as below screenshot :
[image]

check the reason, in line 1678：
https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/ros_filter.cpp#L1678
missing duration time, add timeout to block until find latest  available transform.
will  test and commit later.
Thanks!

ayrton04, 2016-09-21:
This isn't a bug. That transform isn't available immediately when the robot starts, because the *odom*->*base_link* EKF hasn't started producing an estimate yet. As soon as it starts, the error goes away. If you place a blocking timeout in there, then there's a chance that you could be delaying the filter output, which I don't want to do.

I'd rather change the condition by which the error message is produced, e.g., using `ROS_ERROR_STREAM_DELAY`. This will cause the issue to only be reported after some number of seconds.

asimay, 2016-09-21:
The lookupTransform() takes four arguments. The 4th is an optional timeout. It will block for up to that duration waiting for it to timeout.
ok. I check document here:
http://wiki.ros.org/tf2/Tutorials/tf2%20and%20time%20%28C%2B%2B%29
The lookupTransform() takes four arguments. The 4th is an optional timeout. It will block for up to that duration waiting for it to timeout.
So **lookupTransform() will actually block until the transform between the two turtles becomes available** (this will usually take a few milliseconds), OR --if the transform does not become available-- until the timeout has been reached.

I guess this will be effect.

ayrton04, 2016-09-21:
I'm aware of how the fourth argument behaves. The argument is always present, and uses a default value when not specified. What I am saying is that I want to keep the default, and do not wish to change the behavior of that line of code.

The real issue is that I am printing out the error too quickly.

asimay, 2016-09-21:
OK. Thanks. Got it.

closed 2016-09-26 by asimay