cra-ros-pkg/robot_localization #306: Bug report. nav module, use_odometry_yaw=true will cause tf transform error.

asimay, 2016-09-22:
Bug report. 
when set `use_odometry_yaw = true` in navsat_transform_template.launch,  will report `"could not obtain transform from  to base_link, Error was....."`, and the print info loop infinitely. Screenshot as belows:
[image]
after investigating the code, I find the code missing initialization part of imu msg header initialization, when call `lookupTransformSafe()` , parameter `msg->header.frame_id` is null.
at here:
https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/navsat_transform.cpp#L706

module should define `imu_frame_id ` in code:
imu->header.frame_id = "imu_link";
imu->header.stamp = msg->header.stamp;

I modify on my own machine, and test and works well.
Thanks!

ayrton04, 2016-09-22:
Nice catch! Instead of giving it a static string value, however, you need to give it the `base_link_frame_id_` frame, so that when we look up the transform in the IMU callback, we get the identity transform:

```
sensor_msgs::Imu *imu = new sensor_msgs::Imu();
imu->orientation = msg->pose.pose.orientation;
imu->header.frame_id = base_link_frame_id_;
sensor_msgs::ImuConstPtr imuPtr(imu);
imuCallback(imuPtr);
```
Alternatively, you could do this:

```
sensor_msgs::Imu *imu = new sensor_msgs::Imu();
imu->orientation = msg->pose.pose.orientation;
imu->header.frame_id = msg->child_frame_id;
sensor_msgs::ImuConstPtr imuPtr(imu);
imuCallback(imuPtr);
```

asimay, 2016-09-22:
ok. Thanks!
I'm not sure whether to use base_link_frame_id_ or imu_frame_id here.
generally, people will define their own imu frame id. 
and ROS [REP 145](https://github.com/paulbovbel/rep/blob/5b0c670130da8222ab28d5aba75dce5a26442ffe/rep-0145.rst)
define the imu frame id is imu_link..

ayrton04, 2016-09-22:
No, I think you're missing what this code is doing.

We can either get our orientation from the IMU or from the EKF state estimate. If we get it from the IMU directly, then yes, you would want the IMU frame. However, in this case, you are getting the data from the EKF output. The EKF only concerns itself with two frames: `odom` (or map) and `base_link`. Ordinarily, when we transform pose data (like orientations), we need to transform it into the `odom` frame, but IMU data is tricky. Regardless, what we want is to just accept the orientation as-is, and not transform it at all. Since [this line](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/navsat_transform.cpp#L497) is looking up the transform to `base_link`, we need to give it the same frame so that it gets the identity transform.

asimay, 2016-09-22:
ok.  so at here, it is specially let   `targetFrame == sourceFrame` for identity transfrom in lookupTransform(), right?
got it.

asimay, 2016-09-22:
ok. Need my commit for this code change? I'm pleasure to do this.

ayrton04, 2016-09-22:
You got it! Then when we transform the orientation, it'll remain the same. Yes, feel free to submit a PR! That would be very helpful. Thank you!

closed 2016-09-24 by ayrton04