cra-ros-pkg/robot_localization #311: Provide the estimated global pose with the filter state's covariance matrix

corot, 2016-10-18:
Is there any easy way to access the filter state's covariance matrix? If not, would it make sense to publish the estimated global pose as a PoseWithCovarianceStamped when world frame is 'map'?

Now filter publishes nav_msgs/Odometry on odometry/filtered, but when using robot_localization for global pose estimation, the global pose is more meaningful. Yes, we have the tfs, but a PoseWithCovarianceStamped would make visible the filter state's covariance matrix. Like that, it will provide both an easy-to-visualize pose and a convenient stability measure.

Thanks!

ayrton04, 2016-10-18:
The state estimation nodes publish an Odometry message. See the definition of nav_msgs/Odometry. Both the pose and twist data have covariance matrices, both of which are being filled out and published. The pose portion of the Odometry message *is* the global pose.

corot, 2016-10-18:
You are right; odometry/filtered is what I was asking for. Maybe the name is a bit misguiding, but makes sense to use odometry messages to include velocity.
Thanks!

closed 2016-10-18 by corot