cra-ros-pkg/robot_localization #312: Using UKF_node and set 2d pose that have problem

sai5555, 2016-10-19:
This my ukf.yaml
> frequency: 30
sensor_timeout: 0.1
transform_time_offset: 0.0
two_d_mode: true
print_diagnostics: true
map_frame: map
odom_frame: odom
base_link_frame: base_link
world_frame: odom
alpha: 0.001
kappa: 0
beta: 2
odom0: /odom
odom0_config: [false,false,false,
               false, false, false,
               true, false, false,
               false, false, false,
               false, false, false]
odom0_differential: false
odom0_relative: false
odom0_queue_size: 50
odom0_nodelay: false

I confirm the  position.z  is zero  in  /odom messages.
I start  2d pose estimate using ukf  and  pose_z  is non-zero value.
I  try track it and find out when computing sigma point , the position.z provide non-zero value.
` sigmaPoints_[sigmaInd + 1] = transferFunction_ * (state_ + weightedCovarSqrt_.col(sigmaInd)); ` in ukf.cpp

**Is it OK ?**

By the way , I using EKF to estimate pose and position.z is always zero.

sai5555, 2016-10-24:
Could anyone answer my problem , please ?
Thanks

ayrton04, 2016-10-24:
Sorry, I have a day job that does not involve this package. Have you asked a question on answers.ros.org? If so, can you point me to it?

sai5555, 2016-10-24:
Yes , I posted it on answers.ros.org. But nobody reply it.

http://answers.ros.org/question/245981/robot_localization-using-ukf-have-some-problem/

ayrton04, 2016-10-24:
In general, if you don't get a response on answers.ros.org, it means that either nobody has an answer to the question, or they haven't had time to look into it for you. I'll close this issue and respond on the ROS answers site when I get a chance.

closed 2016-10-24 by ayrton04