cra-ros-pkg/robot_localization #314: Expose acceleration state via ROS interface

reinzor, 2016-11-01:
The acceleration state is currently not exposed since the `nav_msgs/Odometry` does not hold the acceleration attribute.

Here: http://www.ros.org/reps/rep-0147.html , people suggested adding an acceleration attribute to the message; however, this would break the current API. An other possibility is exposing the acceleration state via a `geometry_msgs/AccelWithCovarianceStamped` message.

I would like to integrate this into the ROS filter, what do you think about this?

ayrton04, 2016-11-01:
Sounds like a good idea! Should be pretty straightforward. Just so we don't introduce a new topic for everyone, please make publication optional, with the default parameter value set to false.

reinzor, 2016-11-01:
OK thanks, I will send a PR later this week.

-Rein

ayrton04, 2016-11-14:
Merged.

closed 2016-11-14 by ayrton04