cra-ros-pkg/robot_localization #318: Add GetState service to robot_localization

reinzor, 2016-11-04:
Hi,

Currently, it is not possible to query the robot's state at a specific time stamp with use of a service call. With use of`TF` listener, the robot's position can be queried and TF will do interpolation, however, some extrapolation is not possible and will not be added (https://github.com/ros/geometry2/issues/98). Besides this, `TF` will only give you the position, no velocity and/or acceleration state. 

What do you think about adding a service call to obtain the robot's state at a specific time stamp?

Another option would be to make some kind of client that listens to `TF`, `nav_msgs/Odometry` and `geometry_msgs/AccelWithCovarianceStamped`, buffers these and do interpolation/extrapolation. However, all this information is available in the robot_localization node.

Thanks!

ayrton04, 2016-11-07:
So I can see the value in this, definitely, but I'm leaning towards making it another node, for a few reasons:

1. Despite the fact that the information is available inside the state estimation node, the vast majority of the logic will be dedicated to the interpolation and extrapolation. If we make a standalone node, then the only difference will be that we have to add a few subscribers.

1. A new node could conform to ROS coding standards, rather than having to match the existing non-conformant style.

1. I'd prefer not to incur any more overhead in the run loop for the state estimation nodes, if I can. It's already getting to a point where higher frequencies result in frequent warnings about the update taking too long.

rokusottervanger, 2016-11-07:
We're thinking about making a class that listens to the `nav_msgs/Odometry` and `geometry_msgs/AccelWithCovarianceStamped`, buffers those and provides (at least) a method to query the state at a given time. The functionality would be similar to tf's [transformlistener](http://docs.ros.org/diamondback/api/tf/html/c++/classtf_1_1TransformListener.html), but only for a specific transform, and including velocity and acceleration. We could then wrap this in a ROS node that provides a service to query the state. We think this would be a good addition to the robot_localization package. Do you agree?

ayrton04, 2016-11-09:
Absolutely!

reinzor, 2016-11-11:
We are almost finished with the c++ implementation of the RobotLocalizationListener, are you interested in a PR? If so, would a Python object also be desired, just as the TFListener?

ayrton04, 2016-11-11:
I'm fine with just a C++ client, but if you have the cycles to do it, then please feel free.

One note: I may not get to reviewing all of these this week, but I will definitely test and merge them soon.

ayrton04, 2016-11-11:
And thank you once again!

reinzor, 2016-11-15:
PR https://github.com/cra-ros-pkg/robot_localization/pull/324 for the cpp implementation

ayrton04, 2017-09-24:
Merged.

closed 2017-09-24 by ayrton04