cra-ros-pkg/robot_localization #321: Use true course in navsat transform node

reinzor, 2016-11-10:
Some GPS sensors also output a 'true course' which is used by the nmea navsat driver to calculate a velocity vector:

https://github.com/ros-drivers/nmea_navsat_driver/blob/jade-devel/src/libnmea_navsat_driver/driver.py#L159

This information could also be used in the navsat transform node in case an IMU is not available. I would like to implement this functionality but this implies I think a lot of changes in the code. 

My approach would be adding a TwistStamped subscriber of which the orientation is and velocity is extracted. This could be merged in the odom message.

What do you think? 

Thanks,

Rein

reinzor, 2016-11-10:
Or is this where we should use the wait_for_datum functionality for and use only position updates?

reinzor, 2016-11-10:
I am now calling a /datum `setDatum` service with both information as an input:

        q = transformations.quaternion_from_euler(0, 0, atan2(TWIST.twist.linear.y, TWIST.twist.linear.x))
        
        r = SetDatumRequest()
        r.geo_pose.position.latitude = GPS.latitude
        r.geo_pose.position.longitude = GPS.longitude
        r.geo_pose.position.altitude = GPS.altitude
        r.geo_pose.orientation = Quaternion(x=q[0], y=q[1], z=q[2], w=q[3])
        
        rospy.loginfo("Calling setDatum")
        try:
            resp1 = set_datum(r)
            
        except rospy.ServiceException, e:
            rospy.logerr("Service call failed: %s"%e)

Seems also to be a nice solution, should we add some documentation about this ..?

ayrton04, 2016-11-10:
How does it behave when the GPS is still? Would it make more sense to add a check on the velocity magnitude?

In any case, sure, that seems like something worthwhile to put in the documentation.

reinzor, 2016-11-10:
I would say that is up to the user. I can make a python scripts and add this to the repo if desired. I have to add rospy to the deps though.

ayrton04, 2016-11-10:
Nah, I think the documentation is sufficient. Thanks!

ayrton04, 2017-02-21:
What's the status on this? No rush on my end, just want to know whether to keep this open.