cra-ros-pkg/robot_localization #326: Pose and velocity calculation of frames fixed to the robot

rokusottervanger, 2016-11-23:
For our (automotive) use case, we need to calculate the (poses and) velocities of sensors (fixed to the robot) with respect to the fixed world to be able to track their readings in a world-fixed frame of reference. Their poses can be found using a tf listener, but the velocities depend on this transformation as well as the motion of the robot base, which is estimated using nodes in the robot localization package. 
We feel this is something that other people may also need to do. Especially if they have sensors capable of measuring velocity, such as radar. And we think this would be a nice addition to the robot localization package, wouldn't you agree?

ayrton04, 2016-11-23:
Sure. Any reason it can't be rolled into your PR? The `GetState` service could include a [frame_id](https://github.com/cra-ros-pkg/robot_localization/pull/324/files#diff-eee1a9fe3a42f3ef637b5d48825e4acaR1) in the request.

rokusottervanger, 2016-11-23:
Sounds good, I'll add it there.

ayrton04, 2017-09-24:
Merged.

closed 2017-09-24 by ayrton04