cra-ros-pkg/robot_localization #328: Fix issue with basic IO tests

ayrton04, 2016-11-28:
This happens occasionally, and it's annoying:

http://build.ros.org/job/Kpr__robot_localization__ubuntu_xenial_amd64/12/testReport/junit/(root)/InterfacesTest/ImuTwistBasicIO/

SGSSGene, 2017-02-28:
This seems to be an conversion issue.
Simple Code to reproduce:

```
  double r = 0.;
  double p = -M_PI/2.0 -1e10;
  double y = 0.
  Matrix3x3 mat;
  mat.setRPY(r, p, y);
  mat.getRPY(r, p, y);
  // r, p, y are now as expected roughly (0, M_PI/2, 0)
```
if you change p just a little:
```
  double r = 0.;
  double p = -M_PI/2.0 + 1e10;
  double y = 0.
  Matrix3x3 mat;
  mat.setRPY(r, p, y);
  mat.getRPY(r, p, y);
  // r, p, y are now (M_PI, M_PI/2, M_PI)
```

I'm not sure how to fix this unit test to suppress this behavior.

ayrton04, 2017-03-01:
Thanks for the PR!

closed 2017-03-01 by ayrton04