cra-ros-pkg/robot_localization #329: robot_localization high cpu usage and no output

natejgardner, 2016-12-15:
I have an IMU (orientation only) and rtabmap connected to robot_localization ukf. When I include the pose from the IMU and the odom from rtabmap, robot_localization works fine and uses 1-2% CPU. However, when I add the twist messages from the IMU, robot_localization only produces about 10 messages before ceasing to have any output. When I watch it in htop, it uses 100-105% CPU. I assume that whatever it's doing is blocking indicated by no messages being published and ctrl-c failing to work. I examined the twist data going into robot_localization from the IMU and they look fine. They're coming in at about 375 Hz like the pose messages. rtabmap is coming in at about 10Hz with its odom message. Transforms have been applied, but I'm not sure if they are correct in the way they relate the rtabmap reference frame with the IMU frame. Any idea why this would be happening?

ayrton04, 2016-12-15:
Do you have a question for this on ROS answers? If so, can you direct me to it? I've been neglecting it lately, sorry. In any case, this typically points to an issue with one of your covariance matrices. If the filter's estimate error covariance becomes singular (via bad measurement covariances or bad covariance parameters), I find the filter slows to a crawl and stops producing data. Can you post a sample IMU message and your configuration?

natejgardner, 2016-12-15:
Thanks for your fast response! The ROS Answers question is here:

http://answers.ros.org/question/250086/robot_localization-high-cpu-and-no-output/

I will get you a sample of the input data later today. The covariance is
static for both the IMU pose and twist, measured by analyzing noise from
data samples. Since only orientation is available pose covariance is for θ,
φ, ψ, and twist covariance is for dθ, dφ, dψ. All other points in the
covariance matrix diagonal are 0, such that for the pose, only the fourth,
fifth, and sixth elements on the diagonal are occupied and for twist the
tenth, eleventh, and twelfth on the diagonal are occupied. All other values
in the matrix are zero.

In the launchfile, pose is set to [false, false, false, true, true, true,
false, false, false, false, false, false, false, false, false] and twist is
set to [false, false, false, false, false, false, false, false, false,
true, true, true, false, false, false].

I discovered last night that if I set the frequency of robot_localization
to any value <= 2.5 Hz, it will work for a long time. Any value higher and
it can never get more than 11 messages before ceasing to output. It seems
this is the threshold where the CPU runs at 98% or more. Before that, the
CPU never uses more than 6%.

ayrton04, 2016-12-15:
Can you just paste your entire config file for r_l along with the sample IMU messages when you get them? Thanks.

natejgardner, 2016-12-16:
Here is my launch file:

```
<launch>

  <node name="odom_transform" pkg="tf" type="static_transform_publisher"
      args="0 0 0 0.506 0.52 0.08 camera_link base_link 100" />

  <node pkg="covariance_applicator" 
        type="covariance_applicator_node" 
        name="covariance_applicator"
        clear_params="true"
        output="screen">
        <param name="theta" value="0.0000000003302221698534690" />
        <param name="phi" value="0.0000000003302221698534690" />
        <param name="psi" value="0.0000000003302221698534690" />
        <param name="dtheta" value="0.0000000173106564426725" />
        <param name="dphi" value="0.0000000173106564426725" />
        <param name="dpsi" value="0.0000000173106564426725" />
  </node>
        
  <node pkg="robot_localization" 
        type="ukf_localization_node" 
        name="ukf_localization"
        clear_params="true"
        output="screen">
        <param name="frequency" value="30" />
        <param name="sensor_timeout" value="1" />
        <param name="map_frame" value="map" />
        <param name="odom_frame" value="odom" />
        <param name="base_link_frame" value="base_link" />
        <param name="world_frame" value="map" />
        <param name="transform_time_offset" value="0.0" />

        <param name="pose0" value="/covariance_applicator/pose_with_generated_covariance" />
        <param name="twist0" value="/covariance_applicator/twist_with_generated_covariance" />
        <param name="odom0" value="/rtabmap/odom" />

        <rosparam param="pose0_config">
          [false, false, false,
           true, true, true,
           false, false, false,
           false, false, false,
           false, false, false]</rosparam>

        <rosparam param="twist0_config">
          [false, false, false,
           false, false, false,
           false, false, false,
           true, true, true,
           false, false, false]</rosparam>
          
        <rosparam param="odom0_config">
         [true, true, true,
          true, true, true,
          true, true, true,
          true, true, true,
          false, false, false]</rosparam>
        
        <param name="pose0_differential" value="false" />
        <param name="odom0_differential" value="false" />
        <param name="twist0_differential" value="false" />
        <param name="pose0_relative" value="true" />
        <param name="odom0_relative" value="true" />
        <param name="twist0_relative" value="true" />
        <param name="print_diagnostics" value="true" />

        <param name="pose0_queue_size" value="500" />
        <param name="odom0_queue_size" value="500" />
        <param name="twist0_queue_size" value="500" />
[... 284 lines cut ...]
child_frame_id: base_link
pose: 
  pose: 
    position: 
      x: -0.000328303095486
      y: -0.00405944864023
      z: 0.00831629624722
    orientation: 
      x: -0.00177834449936
      y: 0.00575986233378
      z: 0.00294158872596
      w: 0.999977504013
  covariance: [4.198104728050066e-05, 6.615418728299239e-16, 2.0490352662247204e-13, -5.559198662865039e-25, -9.449802356584182e-25, -1.1788658019390678e-24, 6.615418727976121e-16, 4.198104727865665e-05, 6.599358203171804e-14, 1.0340759002913375e-23, 1.9526329896917108e-23, 7.664422307794503e-24, 2.0490352662247204e-13, 6.599358203254522e-14, 4.200740774890077e-05, -1.0921744938415632e-23, -1.9897349504226274e-23, -5.216145569687941e-24, -5.535795519077324e-25, 1.0340758969166712e-23, -1.0915326500101388e-23, 4.1876279739577575e-05, 1.3610871577371871e-15, 2.522134032931778e-13, -9.385131180346818e-25, 1.952633864523477e-23, -1.989573012016568e-23, 1.3610871577662677e-15, 4.187627973585399e-05, 8.124020287308249e-14, -1.182080493639571e-24, 7.658791259702197e-24, -5.216141987586049e-24, 2.5221340329483217e-13, 8.124020287349608e-14, 4.2007380755630305e-05]
twist: 
  twist: 
    linear: 
      x: -0.0262780938663
      y: -0.0453231487448
      z: 0.117106306662
    angular: 
      x: -0.0284828844293
      y: 0.151705849613
      z: 0.0575076826239
  covariance: [0.0016951597397805353, 1.0922074358402892e-15, 7.64538903503356e-15, 2.9940762920356505e-24, -1.8831227842929853e-22, 6.562529429671821e-25, 1.0922074358402778e-15, 0.0016951597397769033, -1.0501152316128596e-14, 4.975407949249108e-25, -3.1288907073229983e-23, 1.0906162530782703e-25, 7.645389035033295e-15, -1.0501152316128609e-14, 0.0027034258895126686, -3.1175162443402456e-24, 1.960343962740558e-22, -6.828429926536926e-25, 2.994994140553549e-24, 4.975407949225095e-25, -3.117236951828786e-24, 0.0006863419042537942, 2.281450470632591e-15, -1.3684808497232686e-13, -1.883147260251118e-22, -3.129013087113633e-23, 1.9603439627400334e-22, 2.281450470635279e-15, 0.0006863419042463856, -1.0566756068404783e-13, 6.562529429042224e-25, 1.0906162518084885e-25, -6.831399635176096e-25, -1.3684808497232681e-13, -1.0566756068405043e-13, 0.0013583201043499756]
```

natejgardner, 2016-12-16:
We found the problem. It had nothing to do with r_l. We were inadvertently publishing bad data on our twist topic. The odom output of r_l was accidentally being routed into the twist message... ROS took the odom data type and evidently interpreted it as a twist message. Yes, our covariance matrix is technically singular, but I believe that's okay due to the values in the boolean matrix. In any case, it now works, and it's totally not a problem with r_l.

closed 2016-12-16 by natejgardner

ayrton04, 2016-12-16:
Cool. One thing you should fix, though, is that your IMU twist data has a `frame_id` of map. That should be changed to base_link (assuming the IMU isn't mounted with none-zero roll or pitch).

ayrton04, 2016-12-16:
Two other things:

1. You have `relative` mode turned on for twist data. Relative mode is only used for pose variables. It doesn't hurt anything as you have it; it's just being ignored.

1. You are fusing orientation from both rtabmap and your IMU. You may see some jitter as it jumps back and forth between their orientation estimates. I'd turn on `differential` mode for whichever one provides the poorer orientation estimate.

Can you please update your ROS answers question?

natejgardner, 2016-12-16:
ROS Answers is closed and links back here. Thanks for noticing those. We caught the frame issue and switched it to base_link and it works much better now. We are noticing the jitter, but rtabmap is jittery anyway. The orientation ends up looking very stable, but due to the lack of translational data from the IMU, position jitters a lot. This is probably partially due to the transform closely linking the IMU to the map. Thanks so much for your help!