cra-ros-pkg/robot_localization #332: Filter seems to ignore orientation

GuidoBartoli, 2017-01-12:
Hi,
I'm currently trying to make robot_localization work in a configuration similar to the dual_ekf_navsat_example.
Before building up the second EKF for global state estimation, I noticed that while fusing wheel odometry with IMU data, EKF output seems to ignore orientation received from IMU, i.e. the resulting orientation is almost always fixed regardless the original orientation received from input.
I thought about a problem in IMU data, so for simplicity I put only wheel odometry inside EKF to see if the orientation would be kept.
This is my configuration for the single EKF input:
>
       <param name="odom0" value="$(arg wheels_odom)"/>
       <rosparam param="odom0_config">
            [ false, false, false,
              false, false, false,
              true,  true,  false,
              false, false, true,
              false, false, false ]
        </rosparam>
        <param name="odom0_differential" value="false"/>
        <param name="odom0_relative"     value="false"/>
        <param name="odom0_queue_size"   value="10"/>
        <param name="odom0_nodelay"      value="false"/>

I'm using **rosbag** to replay experimental data, so my ROS graph is very simple:

[image]

When I display both wheel and filtered odometry in RViz (yellow = wheels, green = EKF), the problem is fairly evident: the green arrows are never aligned along the path (I kept all odometry messages for clearer visualization)

[image]

I'm using the latest github version, do you know what am I missing here?

Thanks,
Guido

ayrton04, 2017-01-12:
Did you ask this question on ROS Answers? If so, can you point me to it? Also, can you please post sample IMU messages and your full EKF config?

GuidoBartoli, 2017-01-18:
Dear Ayrton,
excuse me for not having provided any update, yesterday I found the source of the problem and I forgot to further comment on this issue.

The problem was that wheels odometry was fine (yellow line), but the IMU device CPU had a problem and kept sending zero orientation and that was fooling the Kalman filter.
Below you can see the filter output (green line) against the bare wheels odometry (blue line). The robot started and ended on the same position, so the trajectory was supposed to be a closed circle. The filter output is doing an excellent job on correcting errors and the final positon error (black line) on a total of about 70 meters is only 1.3 meters!

[image]

Thank you very much for your kindness,
Best regards

closed 2017-01-18 by GuidoBartoli

ayrton04, 2017-01-18:
That's good news! Thanks for updating.