cra-ros-pkg/robot_localization #333: Feature: Reset Filter on backwards time jumps

SGSSGene, 2017-01-13:
Hi

I want to add an optional feature that detects that time is jumping backwards and resets the filter. The Motivation is to be able to play a bag and jump forth and back in time.
Since forward jumping is kind of working I only want to focus on the jump backwards in time.

My current Idea is to add a RosFilter<T>::reset() function which is responsible for clearing queue and putting the filters into there initial state.
Assuming that messages on one single topic are being publish in order by time, I could detect a jump back and call the reset() function.
Additional to this I would like to deactivate this feature by default and only activate it if a certain ros parameter is given. This would preserve old and expected behavior.

This would also solve the original problem @hbadino had and described in #163 

Asking for some feedback, going to open a pull request soon

ayrton04, 2017-01-18:
I think you'd have to be careful with this, as measurement timestamps are not guaranteed or assumed to be arriving with monotonically increasing time stamp values. In fact, we recently merged a PR that allows old measurements to be fused. Instead, you may want to keep checking ros::Time::now() when you receive a measurement, and see if *that* jumps backwards. You can also look at the rviz source code to see how they handle it, since rviz does this reliably.

SGSSGene, 2017-01-18:
I assume you are talking about PR #266? This PR only deals with out of order on different topics. When messages out of order arrive on the same topic (more precise same callback like acceleratorCallback). Older messages are just being ignored.

`ros::Time::now()` was also in discussion and I looked into it, but this isn't a fit. The problem here is that after detecting a time jump in `ros::Time::now()` there might be a few messages still coming from the future after I already reseted the filter. This is because of the subscriber and publisher queues.

In (almost) every cases it is save to assume that one single sensor sends monotonically increasing timestamps. Jumps back in time on one single topic are already detected in the current code. All I want to add is a reset() function (that has to be turned explicitly on by params).

ayrton04, 2017-01-18:
> ros::Time::now() was also in discussion and I looked into it, but this isn't a fit. The problem here is that after detecting a time jump in ros::Time::now() there might be a few messages still [...]

You can clear the callback queue. Using `ros::Time::now()` is more explicit and makes the filter not depend on another node's output to determine if the time server has changed. I was mistaken before; it's the `tf` library that automatically does this, and it uses `ros::Time::now()` to make the determination. See line 104 [here](http://docs.ros.org/jade/api/tf2_ros/html/c++/transform__listener_8cpp_source.html).

> In (almost) every cases it is save to assume that one single sensor sends monotonically increasing timestamps. Jumps back in time on one single topic are already detected in the current code. All I want to add is a reset() function (that has to be turned explicitly on by params).

It's true that sensors send monotonically increasing timestamps, but there are no guarantees that they'll be received in that order. Just today there was another issue filed that was (likely) due to this very thing: https://github.com/cra-ros-pkg/robot_localization/issues/336.

SGSSGene, 2017-01-18:
First of all, thank you for helping and dealing with this issue, I appreciate that a lot.

> You can clear the callback queue....

How do I clear the callback queue? The only thing I was able to find was this: http://answers.ros.org/question/66331/clearing-all-messages-on-a-topic which suggest that it is not possible.

>  It's true that sensors send monotonically increasing timestamps, but there are no guarantees that they'll be received in that order. Just today there was another issue filed that was (likely) due to this very thing: #336.

Ros guarantees (at least with default settings), that messages on one topic are received in the order they are send. (This doesn't apply for messages on different topics.). 
One more thing when I was talking about monotonic I meant weakly increasing monotonic, so having duplicated timestamps is possible.

SGSSGene, 2017-05-26:
fixed by #340

closed 2017-05-26 by SGSSGene