cra-ros-pkg/robot_localization #334: How to use robot localization/ Tutorial?

kenisvery, 2017-01-14:
Hi,

I am trying to implement this to the parrot bebop drone to fly an autonomous flight. How could I use this package into the bebop autonomy? Is there any tutorial for this?

ayrton04, 2017-01-18:
I'm afraid I don't have tutorials for integration with specific robots, as I don't have access to very many. I have used this package with a Parrot Drone, but I had to modify the Parrot driver to output standard ROS message types (IMU and Odometry). Looking at the [documentation](http://bebop-autonomy.readthedocs.io/en/latest/reading.html) for the Bebop driver, it looks like it outputs standard ROS message types, so this package should work.

ayrton04, 2017-01-18:
At the very least, you'd have to disable the `publish_odom_tf` in the Bebop driver, otherwise it will conflict with the EKF.

closed 2017-02-21 by ayrton04

cyberphantom, 2018-05-07:
@ ayrton04 I'm planning to use this package with my ARDrone. You mentioned that you had to make some modifications (IMU and Odometry) to the ardrone_driver. Can you please share these modifications?
I'm planning to use a simulation on Gazebo first then I will move to test it on ARDrone. I'm planning to fuse IMU sensor readings and cmd_vel command? Is there any example launch file? 

Thank You,

ayrton04, 2018-05-07:
I'm afraid I made those changes ages ago when I was working for another company. I don't have them. As I recall, at the time, the AR Drone package was outputting customized message types for sensor data, and I just made it output standard ROS `nav_msgs/Odometry` and `sensor_msgs/Imu` messages instead.