cra-ros-pkg/robot_localization #335: SetPose service hangs (sporadically) and CPU usage hits 100%

spmaniato, 2017-01-15:
Hi all,

We (@bhavyangupta and I) are facing the issue described below. We have done some detective work on our side but haven't been able to determine a cause (neither on our side nor something specific in `robot_localization`) Please take a quick look 🙂 Thanks!

### Issue Description

We are using the EKF node from `rl` and are sporadically observing an issue with the `robot_localization/SetPose` service on `/set_pose`. Specifically, on some occasions that _we cannot replicate_, the service will not respond, causing the calling code to hang. At that point, the EKF node also stops publishing messages on `/odom`. In addition, `top` shows its CPU usage maxing out (normally its usage is around 5-6% and only rarely does it go up to 19% or so):

```
PID USER          PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
860 <user>        20   0   83404   3808   3100 R  97.5  0.4   3:18.65 ekf_localization
```

We dug into the source code (`RosFilter<T>::setPoseCallback`) but didn't find anything obviously suspicious. For example, there's a `while` loop that clears a queue ([ros_filter.cpp#L1826-L1829](https://github.com/cra-ros-pkg/robot_localization/blob/indigo-devel/src/ros_filter.cpp#L1826-L1829)) as well as a bunch of `while` loops elsewhere, but none look like they would run indefinitely 🤔 

### System Setup

This is all taking place on a single-board computer (`armv7hf`) running ROS Indigo for ARM. The `robot_localization` package is installed from Debian packages (i.e. `ros-indigo-robot-localization`). I believe we've seen the issue on both `2.3.0-0trusty-20160728-151324-0700` and `2.3.1-0trusty-20161031-053932-0700` versions (but I'll double-check). We haven't tried a from-source installation yet.

### EKF Configuration

The EKF is taking in wheel odometry and IMU data (of which only the gyro rate is used):

```yaml
two_d_mode: true
frequency: 20
base_link_frame: 'base_link'
world_frame: 'odom'
odom_frame: 'odom'

odom0: 'odom'
odom0_config: [false, false, false,
               false, false, false,
               true, true, false,
               false, false, true,
               false, false, false]

imu0: 'imu'
imu0_config: [false, false, false,
              false, false, false,
              false, false, false,
              true, true, true,
              false, false, false]
```

ayrton04, 2017-01-15:
I think it might be a bit more helpful if you could show me some sample service calls that are causing it to freeze. I'd like to see the values you're sending to the filter.

Also, would it be possible for you to attempt to replicate this issue on a non-ARM board?

spmaniato, 2017-01-15:
Hi @ayrton04 , thanks for the quick response!

1.  > I'd like to see the values you're sending to the filter.

I've posted the service call details below. I'll get us some wheel odom and IMU data as well soon.

```cpp
void ResetEkf()
    {
    robot_localization::SetPose ekf_reset;
    ekf_reset.request.pose.header.frame_id = "odom";
    ekf_reset.request.pose.header.stamp = ros::Time::now();
    ekf_reset.request.pose.pose.pose.position.x = 0;
    ekf_reset.request.pose.pose.pose.position.y = 0;
    ekf_reset.request.pose.pose.pose.position.z = 0;
    ekf_reset.request.pose.pose.pose.orientation.w = 1;
    ekf_reset.request.pose.pose.covariance = boost::assign::list_of(1e-9)(0)(0)(0)(0)(0)
                                                                   (0)(1e-9)(0)(0)(0)(0)
                                                                   (0)(0)(1e-9)(0)(0)(0)
                                                                   (0)(0)(0)(1e-9)(0)(0)
                                                                   (0)(0)(0)(0)(1e-9)(0)
                                                                   (0)(0)(0)(0)(0)(1e-9);
    ekf_reset_client_.call(ekf_reset);
}
```

2.  > would it be possible for you to attempt to replicate this issue on a non-ARM board?

Unfortunately not 😞

spmaniato, 2017-01-16:
@ayrton04 we were able to _artificially_ replicate the issue by running the EKF (same config as above) and then using the simple Python node below to send it _ridiculous_ IMU messages. When the problem manifested, the CPU usage maxed out (see `top` output below) and the last `SetPose` service hang.

On our side, we are now focusing our attention on our IMU pipeline. It could be that our messages have sporadic large values and that one of those happens to hit while the `SetPose` request is being processed. We are checking bagfiles, etc. I'll post another update if we find something noteworthy.

```
top - 00:14:25 up 7 min,  4 users,  load average: 1.10, 0.69, 0.33
Tasks:   1 total,   1 running,   0 sleeping,   0 stopped,   0 zombie
%Cpu(s): 28.4 us,  2.0 sy,  0.0 ni, 69.5 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem:    947740 total,   358372 used,   589368 free,    25016 buffers
KiB Swap:  1048572 total,        0 used,  1048572 free.   159416 cached Mem

PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                            
3878 <user>   20   0   91504   9180   8264 R 101.2  1.0   1:03.81 ekf_localization 
```

```python
#!/usr/bin/env python

import rospy
import sys
from sensor_msgs.msg import Imu
from robot_localization.srv import *

set_pose_srv = rospy.ServiceProxy('/set_pose', SetPose)

def callback(event):
    rospy.loginfo("Set_pose Callback")
    try:
        reset_pose_req = SetPoseRequest()
        reset_pose_req.pose.header.stamp  = rospy.Time.now();
        reset_pose_req.pose.header.frame_id = 'odom'
        reset_pose_req.pose.pose.pose.orientation.w = 1.0
        reset_pose_req.pose.pose.covariance = [0.1] * 36
        set_pose_srv(reset_pose_req)
    except rospy.ServiceException as exc:
        print("srv not found")

if __name__=='__main__':
    topic = sys.argv[1] 
    rospy.init_node('infinite_imu')
    rospy.wait_for_service('/set_pose')
    rospy.Timer(rospy.Duration(3.0), callback)
    imu_pub = rospy.Publisher(topic, Imu, queue_size=1, latch=True)
    imu_msg = Imu()

    imu_msg.header.frame_id = 'imu_link'
    imu_msg.orientation_covariance = [0.1] * 9
    imu_msg.angular_velocity_covariance = [0.1] * 9
    imu_msg.angular_velocity.x = float(10000000000)
    imu_msg.angular_velocity.y = float(10000000000)
    imu_msg.angular_velocity.z = float(10000000000)
    imu_msg.angular_velocity_covariance = [1e-3] * 9
    imu_msg.linear_acceleration.x = float(10000000000)
    imu_msg.linear_acceleration.y = float(10000000000)
    imu_msg.linear_acceleration.z = float(10000000000)
    imu_msg.linear_acceleration_covariance = [10] * 9
    loop_rate = rospy.Rate(10)
    while not rospy.is_shutdown():
        imu_msg.header.stamp = rospy.Time.now()
        imu_pub.publish(imu_msg)
        loop_rate.sleep()
```

ayrton04, 2017-01-18:
Yeah, when I've seen the filter lock up previously, it's pretty much always been a case of a singular matrix somewhere. Sometimes it's because the user specifies a bad initial covariance matrix, but that causes it to lock right away. Other times it's because measurements have bad covariance matrices. Let me know what you find.

spmaniato, 2017-01-19:
@ayrton04 The plot thickens 🤔 According to our bagfiles, the EKF was already in a "problematic" state when our code made the `SetPose` service request. It's just that calling the service made the issue obvious because it hang the calling code. By problematic I mean that it had already stopped publishing `odom` and broadcasting the `odom --> base_link` transform.

We suspect that the root cause (maybe a malformed IMU message), took place before `rosbag record` started 😞 Will keep investigating.

ayrton04, 2017-01-19:
I guess just record all the inputs *and* the filter output until you see it happen again? I'm interested to see how the covariance is growing in your filter output as well.

spmaniato, 2017-01-20:
@ayrton04 We can quite confidently confirm that, in our case, the root cause was a "messed up" IMU message published way before `SetPose` was requested. By "messed up" I mean that some of the message's fields (e.g. `angular_velocity` components) had 38 digits long values 😮 

Feel free to close this issue and thanks again for your help 🙂

ayrton04, 2017-01-21:
Whoa! You need to slow that robot down. :) Thanks for the update!

closed 2017-01-21 by ayrton04