cra-ros-pkg/robot_localization #337: Feature: Detect if time stamps of sensor messages are far behind wall time

sharst, 2017-01-18:
It would be nice if the diagnostics of robot_localization would print a warning if one of the incoming sensors messages lags behind _a lot_. 
Particularly, I had a problem stemming from a clock issue where timestamps of some sensor messages were erroneously set to 0. As the filter can integrate readings from the past, my guess is that they were included but didn't affect the current filter reading in any way. I had a hard time finding out why my sensor readings were seemingly ignored completely.

ayrton04, 2017-01-18:
It's certainly doable, but I'm not sure what is a good way to define "a lot." I also waffle a bit on how much onus should be on the user to make sure their data streams are correct, but in the end, if the node can detect an issue with sensor data and report it, it should.

sharst, 2017-01-18:
For starters, I would suggest that any message with Time 0 looks fishy and there could be a warning. I ran into this issue while playing back data from a bag and fusing it with other data I generated on the fly. Due to an error in rqt_bag, the clock wasn't published correctly and the newly generated messages would all have timestamp 0, while the timestamps of the played back messages were all okay.
This is a very special case and also due to a bug, but it seems to me that timing issues might be common when operating on played back data and therefore a warning (e.g. in the diagnostics) might be helpful.

ayrton04, 2017-02-01:
Agreed.

ayrton04, 2017-12-07:
I think this was addressed by #340? Let me know if you find otherwise.

closed 2017-12-07 by ayrton04