cra-ros-pkg/robot_localization #338: navsat_transform_node rounding datum lat lon

mikehosmar, 2017-01-24:
I have specified a fairly accurate ppp datum of [43.9406655, -078.8942039, 0] but I found out later that the node is actually using  (43.940700, -78.894200, 0.000000). In UTM this isn't insignificant (something like 3~4 meters).  Not sure if this can be fixed but I think it's happening near the line I linked.

[navsat_transform.cpp L124](https://github.com/cra-ros-pkg/robot_localization/blob/a7f8bd92489ad4c76fbb5f63f58b22780cde6a11/src/navsat_transform.cpp#L124)

P.S. Are there any other positions that the position would be rounded possibly in the UTM calculation? I need to be able to be able to compare the gps odometry to other UTM measurements Ive made.

ayrton04, 2017-02-01:
Yeah, that's no good. I wonder if parameter reading is reading that type as a float (32-bit), or if the input stream operator also respects things like `std::setprecision`. Thanks for the heads-up.

mikehosmar, 2017-02-01:
`std::setprecision` sound right. The default precision is what I'm seeing.  So maybe just a `std::setprecision(10)` would be a good fix.

ayrton04, 2017-02-02:
Care to test it and let me know if it works?

ayrton04, 2017-12-07:
Fixed in #391.

closed 2017-12-07 by ayrton04