cra-ros-pkg/robot_localization #345: tf transform jitters when stopping motion

roboticsist17, 2017-02-20:
Hi there,

I'm cross-posting this from ROS Answers, so apologies for the duplicate. It's been a few days though so I'm not sure which forum is best. Thank you very much for your time and insights!

We're trying to use `r_l` on an outdoor ground vehicle with an IMU affixed to the top, and two RTK GPS receivers fore and aft. The IMU provides roll and pitch at 10 Hz (as `imu0`), and the GPS provides relative position (meters from base station) and yaw, and optionally pitch, at 10 Hz (as `pose0`). We don't have any wheel encoders, so we don't have a typical odometry source. As a result, we simply use a static transform from `map` to `odom`, and we are using `r_l` to generate a transform from `odom` to `base_link`.

`r_l` seems to provide a very accurate transform most of the time (judged by visual inspection of rviz), but when the vehicle stops the estimated position of `base_link` jitters (or oscillates) for 1-2 seconds by 0.1-0.5 meters primarily along the vehicle's X axis, then stops at the correct position. However, if we remove `imu0` as an input into `r_l`, the transform stops smoothly and quickly (as does the physical vehicle).

This jitter occurs even when the following are true:

 - The IMU and GPS do not provide competing measurements for the same axes of motion (i.e., we remove pitch from the GPS input)
 - The covariance matrix diagonal on the IMU is temporarily set to `1e12` (for debug purposes), which we would imagine would cause the `r_l` filters to more or less ignore the IMU input
 - The IMU data are all temporarily set to `0` (for debug purposes), and only an irrelevant axis (e.g., `Z''`) is configured to be active

An example GPS message and IMU message are below, as is the YAML configuration for `r_l`.

Any thoughts would be much appreciated! Thank you!

`imu0`:

    ---
    header: 
      seq: 4
      stamp: 
        secs: 1487210998
        nsecs: 225980997
      frame_id: base_link
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
    orientation_covariance: [1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0]
    angular_velocity: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular_velocity_covariance: [1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0]
    linear_acceleration: 
      x: 0.0
      y: 0.0
      z: 0.0
    linear_acceleration_covariance: [1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 1000000000000.0]
    ---

`pose0`:

    ---
    header: 
      seq: 2395
      stamp: 
        secs: 1487211196
        nsecs: 500669002
      frame_id: odom
    pose: 
      pose: 
        position: 
          x: -3.85235159044
          y: -5.17830684957
          z: 0.00309827089337
        orientation: 
          x: 0.00792342224031
          y: -0.0124397107908
          z: 0.53716653655
          w: 0.843347250536
      covariance: [0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.04, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-12, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.04, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01]
    ---

`r_l` YAML configuration:

    robot_localization:
      map_frame: /map
      odom_frame: /odom
      base_link_frame: /base_link
      world_frame: /odom
    
      frequency: 25
      sensor_timeout: 0.11
    
      pose0: /gps_receiver/base_link
      pose0_config: [True, True, True,
                     False, False, True,
                     False, False, False,
                     False, False, False,
                     False, False, False]
    
      imu0: /effector/imu/cab
      imu0_config: [False, False, False,
                    False, False, False,
                    False, False, False,
                    False, False, False,
                    False, False, True]

ayrton04, 2017-02-21:
Yeah, ROS Answers is the preferred method for questions; this is really for bugs or feature requests. My spare time is very limited right now, hence my lack of response on ROS Answers (apologies!).

ayrton04, 2017-02-21:
Will handle this in ROS Answers. If it's a bug, I'll reopen.

closed 2017-02-21 by ayrton04